Bodmer/TFT_eSPI #1: Compile error

MarcFinns, 2017-03-08:
Hi Bodmer,
Not sure if I made a mistake, but after installing the lib the example does not compile - see below.
I did edit the User_Setup.h (as I did in you older library) but i get the error below. It appears as the zip from GitHub has a patch problem. Is there any additional configuration step required? thanks for your time!

/Users/Marc/Documents/Arduino/libraries/TFT_eSPI-master/TFT_eSPI.h:27:46: fatal error: TFT_Drivers\ILI9341_Defines.h: No such file or directory
      #include <TFT_Drivers\ILI9341_Defines.h>
                                              ^
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

pipi61, 2017-03-09:
Hi!
in win81x64 Arduino169_espver230, I try several examples(Wemos D1 mini + ILI9341 2,8" 320x240), and works, really fast
Great works!

MarcFinns, 2017-03-09:
Now that I look at it again with a fresher mind,  I suspect the issue is the backslash in the path. This fails on unix

MarcFinns, 2017-03-09:
Confirmed. The lib uses hardcoded paths with backslash (\) all over. This works on windows but fails on Mac and Linux, that require forward slash (/).
Any chances to conditionally define them? Otherwise at every upgrade a tedious selective search & replace is necessary...
Other than that, a great piece of work!

Bodmer, 2017-03-09:
@MarcFinns 
Thanks for pointing this out, have changed \ to / in the file paths so it should work for both Windows and Linux now!

closed 2017-03-09 by Bodmer

MarcFinns, 2017-03-09:
Thanks! Great lib by the way!