Bodmer/TFT_eSPI #3: WORKS - MISO pin as D/C !!

MarcFinns, 2017-03-09:
Hi Bodmer, further to our previous conversation on using the MISO pin, I did manage to make it work as D/C. 
Basically, only one additional line of code is needed, to set the pin as output after SPI initialisation!

void TFT_eSPI::init(void)
...
  _SPI->begin(); // This will set MISO to input
...   
 pinMode(D6, OUTPUT);

Any chance to add this option in the config file?
Clearly this works if input functions are not used (and ifSPI transactions are not enabled).
thanks!

Bodmer, 2017-03-10:
OK, that is interesting.
For pin constrained applications it is a good solution. I will add this as an option but it does have some implications for the functions that read TFT pixels, so that will have to be made clear to users. A call to read the TFT registers will have to be blocked in the library as I suspect it will cause the pin to revert to being an input. I will do some tests to see what the implications are and if there are any "gotchas".

Thanks for posting this information.

closed 2017-03-10 by Bodmer

MarcFinns, 2017-03-10:
Acknowledged, that's all correct. However in some cases you are exactly short of one pin... and this breaks the project. For example in my case i am not reading from the display and I desperately needed one input. This solves it!

Bodmer, 2017-03-11:
Agreed, I have added your suggestion, can you test it is OK for your application?

Once it is confirmed to work then I will detail the option in the User_Setup file with the appropriate warnings.

Thanks.

MarcFinns, 2017-03-11:
Happy to do so. Let me know when ready. 
I would suggest to make it generic, ( e.g. miso_not_used) rather than linking it to a specific use. In this way it can be used for d/c or cs as well as for other non-display purposes.

MarcFinns, 2017-03-26:
Hi Bodmer, are you still thinking of implementing this suggestion..? please let me know. thanks!

Bodmer, 2017-03-28:
Hi, yes, it was added 17 days ago. I tested it and it works fine. It is just a case of setting DC or CS to output after SPI.begin() inside init() which only needs to be conditional on the pin being defined.  Since then I have made a few more tweaks.

MarcFinns, 2017-03-29:
it is not in the repository though!

Bodmer, 2017-03-29:
This is the [code section and lines](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L183-L193) that set MISO to output.

MarcFinns, 2017-03-30:
ok, I see it now. Will check. Thanks!

MarcFinns, 2017-05-09:
it works, thanks

MarcFinns, 2017-05-09:
Hi Bodmer, Off topic: Is there documentation or a tutorial on the fonts subject + your extensions?
it is rather obscure with built-in, free, proportional etc. Looks like layers of history. Perhaps you have a good pointer? thanks!

Bodmer, 2017-07-15:
Issue #26 may interest you if you need to save more pins.