Bodmer/TFT_eSPI #7: Trouble compiling with User_Setup.h as written

ChiefPoints, 2017-04-10:
First off, thanks for this great library. I plan on making quit a bit of use of it in the near future. All of the ILI9341 demo code works flawlessly on a bare ESP8266-12E, once I got it to compile.
Just to let you know that I had to remove the "D" from the pin number assignments in the User_Setup.h to get the code to compile. I'm not sure if it's an IDE version problem or what, I'm on 1.6.7 with the latest ESP8266 boards definition.

danriches, 2017-04-10:
I'm pretty sure that's a Wemos board designation, I had to change mine too for a bare ESP12-E and mine worked flawlessly after that too. Many thanks Bodmer for this fantastic library it's the fastest I've come across and makes my project look great rather than a bad implementation as it did with any other library. Congrats on your sterling effort!!

ChiefPoints, 2017-04-10:
Yes, it's the fastest I've used as well. I finally gave me a reason to dust off that pile of 12E's I've been sitting on. I'll be the first to admit that I'm no coder, but I found I also had to change the weatherstation example code quite a bit to get it to compile. I had to move all of the "void <function>"'s up higher in the code, as they weren't being read prior to calling the function. Again, it's possible this a an IDE version issue. I'm not sure if anyone else experienced the same issues.

So far these are the only two problems I've found. As the User_Setup.h is populated with more board settings, it will become even easier to use. Kudos on all the great work. :1st_place_medal: 
Really glad I read about it on Hackaday.com :)

Bodmer, 2017-04-12:
Thanks for you positive comments.  I see the Hackaday post has created a lot of interest.

The IDE I am using is the Arduino 1.8.1 with version 2.3.0 of the ESP8266 boards package.

I have tested the library with the NodeMCU 1.0  (ESP-12E module based) boards for convenience and they use the Dx pin designations. If I select the "WeMos D1 R2 & mini" it compiles fine and upload/runs on a NodeMCU 1.0 board so I assume the WeMos and NodeMCU are compatible. What IDE boards were selected to get the pin number errors?

I know some versions of the IDE had problems if functions were not declared before setup(), however I have not had any trouble with IDE 1.8.1.

ChiefPoints, 2017-04-12:
I'm using the "Generic ESP8266 Module" board setting, I've got some nodemcu boards around here somewhere and I'll test those, but I imagine it will compile just fine for them. I did test the newer version of the arduino IDE and that does resolve the issue with having to move the "void <function>"'s higher up in the weather station code. I guess they improved that since I updated last.

Bodmer, 2017-04-12:
Needing function definitions before setup() was a bug in the IDE for a while.

The Dx mapping to GPIOx for the NodeMCU is as follows:

static const uint8_t D0   = 16;
static const uint8_t D1   = 5;
static const uint8_t D2   = 4;
static const uint8_t D3   = 0;
static const uint8_t D4   = 2;
static const uint8_t D5   = 14;
static const uint8_t D6   = 12;
static const uint8_t D7   = 13;
static const uint8_t D8   = 15;
static const uint8_t D9   = 3;
static const uint8_t D10 = 1;

The "Generic ESP8266 Module"  pin mapping seems to miss out pins so perhaps changing to a different module in the IDE will help?

closed 2017-04-12 by Bodmer

Bodmer, 2017-04-14:
The work around now in the library is to re-define the pins so they can be identified in the User_Setup files with the PIN_Dx format, eg:

#define TFT_DC   PIN_D3  // Data Command control pin

This also means that the pin definitions are valid no matter what ESP8266 board is selected. These are the definitions, they are in the User_Setup_Select.h file for convenience.

#define PIN_D0 16
#define PIN_D1 5
#define PIN_D2 4
#define PIN_D3 0
#define PIN_D4 2
#define PIN_D5 14
#define PIN_D6 12
#define PIN_D7 13
#define PIN_D8 15
#define PIN_D9 3
#define PIN_D10 1