Bodmer/TFT_eSPI #8: Trouble using library with DC pin set to D4 instead of D3 -(I found work-around)

bmillier, 2017-04-10:
First off- once I got this working, I see this is an amazingly good driver/example set. Thanks a lot!

My Adafruit ESP8266 Huzzah has D3 (arduino Pin0) dedicated to a programming pushbutton- isn't pinned out. Therefore, I have connected TFT pin DC to D4 (arduino #2 on huzzah). In User_setup.h I have the following line
#define TFT_DC   D4  // Data Command control pin

This doesn't work at all (and I can see the red LED on D3 still flashing) . But, if I change line 78 in TFT_eSPI.h
  from this - #define DC_C GPOC=dcpinmask
to this-  #define DC_C digitalWrite(D4, LOW);
it works fine.

 I also notice that I can't use TFT_DC in place of D4, above, or it won't work.

Alternately, if I change Lines 160, 161 in TFT_eSPI.cpp, as follows:
  dcport    = portOutputRegister(digitalPinToPort(D4));  //was TFT_DC
  dcpinmask = (uint32_t) digitalPinToBitMask(D4);        // was TFT_DC 

AND restore Line 78 in TFT_eSPI.h to its original, this also works OK.
In an example sketch, if I insert 
Serial.println(TFT_DC)    it prints 0 to serial port
but Serial.println(D4) it prints 2, which is the correct Arduino pin# for D4. 
I don't know why, but the #define TFT_DC  D4  doesn't seem to be making its way to the applicable code in the .cpp and .h files.

Bodmer, 2017-04-12:
OK. I do not have a Huzzah board to try. I can only assume the GPIO to virtual pin number mapping is different to a NodeMCU whch is the ESP8266 board that I am using.

I will investigate further.

bmillier, 2017-04-12:
Thanks. On Adafruit's site they give a table relating their pin labelling (Arduino convention) to the nodeMCU convention. So, I am using that to match the nodeMCU definitions that you use. Everything works great if I use D4 in the code, instead of TFT_DC, which is defined as D4 in my version of user setup file. The fact that serial.println sends out 0 for TFT_DC and 2 for D4 (ie Arduino pin2 which is NodeMCU D4)  explains why I can't get the display to work when I use TFT_DC in my version of the setup files.

Bodmer, 2017-04-13:
As a temproary fix the NodeMCU to GPIO pin numbers for the Huzzah can be defined by adding the following lines to the end of the "User_Setup_Select.h" header file in the library:

`//The Dx mapping to GPIOx for the NodeMCU is as follows:
#define D0 16
#define D1 5
#define D2 4
#define D3 0
#define D4 2
#define D5 14
#define D6 12
#define D7 13
#define D8 15
#define D9 3
#define D10 1
`
D0 (GPIO 16) is a bit peculiar as a pin mask cannot be used, so it is best to avoid that pin.

Bodmer, 2017-04-13:
Hi,
I have looked inside the boards package and the Dx pins are not defined for the Huzzah.

I suggest that you select the board as being NodeMCU 1.0. If you do this then no changes to the library will be needed, but perhaps just a small change to your sketch if you use the LED.

The different between the Huzzah and the NodeMCU appears to be just that the LED is on GPOI 0 on the Huzzah but GPIO 16 on the NodeMCU.  So if your sketch uses the LED just use pin 0 (or D3) for that.

bmillier, 2017-04-13:
Thanks for the reply
I was already selecting the nodeMCU as my board because your examples wouldn't even compile when I selected the Adafruit huzzah- for the reason you outline. My issue is that if you just define TFT_DC = D4 in the user_setup file, that definition is not recognized in your other driver files- resulting in those other files thinking that TFT_DC = 0 - which works ok for you, as you are using  port pin 0 in your setup, but not if you need to use any other pin for DC. Try moving the DC pin to some other pin on your nodeMCU , and then defining that  change as you show in the user setup file. I don't think it will work on the nodeMCU as it  doesn't on the huzzah.
 I think that TFT_DC has to be defined as GLOBAL or you have to modify those lines in the code as I described in my original post.

Brian Millier
Computer Interface Consultants
31 Three Brooks Dr.
Hubley N.S.
B3Z1A3
(902) 876-8645
Cell (902) 717-5834

bmillier, 2017-04-13:
I am using Arduino 1.80 - I should have mentioned that. And I am editing the various h and cpp files using the  Visual micro plug in for Visual studio.

Brian Millier
Computer Interface Consultants
31 Three Brooks Dr.
Hubley N.S.
B3Z1A3
(902) 876-8645
Cell (902) 717-5834

Bodmer, 2017-04-14:
OK, I have reopened this issue, it seem I got lucky!

Bodmer, 2017-04-14:
This seems to be a bug in the IDE that only manifests itself in certain conditions.  The conditions where this happens is not clear.

The work around now in the library is to define and that identify the pins in the User_Setup files with the PIN_Dx format. This also means that the pin definitions are valid no matter what ESP8266 board is selected.  These are the definitions, they are in the User_Setup_Select.h file for convenience.

#define PIN_D0 16
#define PIN_D1 5
#define PIN_D2 4
#define PIN_D3 0
#define PIN_D4 2
#define PIN_D5 14
#define PIN_D6 12
#define PIN_D7 13
#define PIN_D8 15
#define PIN_D9 3
#define PIN_D10 1

closed 2017-04-14 by Bodmer

bmillier, 2017-04-14:
Problem is fixed now - thanks.