Bodmer/TFT_eSPI #9: TFT_eSPI.h: swap() clashes with std::swap

bisko, 2017-04-11:
I'm building a geo points map renderer for a project of mine and while trying to sort the vectors of points, I stumbled upon a strange issue.

The `swap()` method, defined in `TFT_eSPI.h` clashes with `swap` from the standard `utility` library. 

Here's the error:

`error: call of overloaded 'swap(GeoPoint&, GeoPoint&)' is ambiguous
swap(*__a, *__b);`

It wasn't very obvious why it was failing from the code, but it seems that `std::sort` is using `swap` in its implementation.

Here's the relevant code that causes the issue:

```
typedef std::vector<GeoPoint> GeoPointList;

...

GeoPointList pointsList;

...

struct GeoPointComparator {
    inline bool operator() (const GeoPoint& struct1, const GeoPoint& struct2) {
        return (struct1.distance < struct2.distance);
    }
};

...

void sortList() {
    std::sort(this->pointsList.begin(), this->pointsList.end(), GeoPointComparator());
}
```

The above are code excerpts that cause the issue. My C++ is not at a very high level, so please forgive any non-optimal syntax :)

In my case, if I renamed `TFT_eSPI.h`'s `swap` to `swap2` for example, the code compiles and runs without any issues.

It would be awesome if this method could be renamed to something that is scoped down to only the `TFT_eSPI` library, so it doesn't clash with the standard library.

Bodmer, 2017-04-12:
OK, thanks for this. I will fix this as you suggest.

Bodmer, 2017-04-12:
Done.

closed 2017-04-12 by Bodmer