Bodmer/TFT_eSPI #10: Not working with ILI9163 128x128 display?

larsenglund, 2017-04-14:
Hi! I'm trying to use this library with a 128x128 ILI9163 display. The display works fine with this library: https://github.com/sumotoy/TFT_ILI9163C. I have set `#define ILI9163_DRIVER` in User_Setup.h and I am overriding the width in the constructor with `TFT_eSPI tft = TFT_eSPI(128, 128);`. I am using the following pin defines:
```
#define TFT_CS   15  // Chip select control pin
#define TFT_DC   0  // Data Command control pin
#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
```
And GPIO12 (MISO) is unconnected since the display does not have that pin.
I have tried a couple of the examples but still I only get a white display. Any thoughts on what might be the problem?

Bodmer, 2017-04-15:
A white screen means the TFT is not being configured correctly at start up.

Do you mean you power the display at the start of setup?  If so try waiting for say 200ms after switching on the display before trying to initialize it.

I had a quick look at the sumotoys library, I don't see #include "../_display/TFT_ILI9163C_RED_PCB_OLD.h" in _settings/TFT_ILI9163C_settings.h) so I can't check the config settings and see if mine are the same.  If you do get it running it should run a bit faster than the Teensy 3 performance figures I see are quoted.

larsenglund, 2017-04-15:
Thanks for the quick reply!
Yes I power the display on at the start of `setup()` (using a mosfet) and I have tried adding 1 second delay after (even though It's not needed when using sumotoys library).

Sorry, forgot to mention that I use the Pre-Release-1.0p7 branch. The header is available here:
https://github.com/sumotoy/TFT_ILI9163C/blob/Pre-Release-1.0p7/_display/TFT_ILI9163C_RED_PCB_OLD.h

Bodmer, 2017-04-15:
OK, thanks.
I had a look at the configuration file and it is different to the one used in the TFT_eSPI library.

So it looks like the TFT_eSPI library will not work with your particular display setup.  This situation may change in the future but at the moment I do not have a similar display to develop and test.

larsenglund, 2017-04-16:
Hmm, ok, how would you go about adapting to the display if you had one? I could give it a try but could use som pointers. I'm guessing it's more to it than taking your ILI9163-config and exchanging some register definitions..

Bodmer, 2017-04-16:
All the TFT driver specific stuff is in the TFT_Drivers folder, for each chip designation there are 3 files. X_defines which set up register names etc, X_init which is the display initialisation code run at startup and X_rotation which handles the register changes and offsets needed when the diplay orientation is changed.

So those are the key files to adapt.

I have updated my copy of the library recently to add support for a ST7735 128x128 display I will upload to Github later today.  This will handle the offsets needed for certain display rotations If I have time I will make a "best guess" update for your display for you to try, based on the sumotoy code, but this will be untested.

larsenglund, 2017-04-16:
Awesome, I hope you do get the time, looking forward to trying it out!

Bodmer, 2017-04-18:
Hi can you try 2 things:
1. Load the Read_ID_bitbash sketch in the ILI9341 example set, change the pin settings in the sketch and report the output seen on the Serial Terminal (115200 baud)
2. Try the S6D02A1 driver (typically this works OK with an ILI9163C)

larsenglund, 2017-04-18:
Hi! Read ID gives me this:
```
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
TFT driver register values:
===========================
Register 0x01: 0x00
Register 0x04: 0x548066
Register 0x09: 0x4510000
Register 0x0A: 0x08
Register 0x0B: 0x08
Register 0x0C: 0x05
Register 0x0D: 0x00
Register 0x0E: 0x00
Register 0x0F: 0x00
Register 0x2E: 0x00
Register 0x3F: 0xFFFFFFFF
Register 0xDA: 0x54
Register 0xDB: 0x80
Register 0xDC: 0x66
===========================

Looks like driver chip is: ILI9163C (based on datasheet ID)
```
The S6D02A1 driver gives me a dark grey display on the same examples as I tried before that gave me a white screen.

Bodmer, 2017-04-18:
I have made a minor change to the [file here ](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9163_Init.h#L15)so sleep mode exit is enabled.

Can you test it with this updated file?

larsenglund, 2017-04-18:
It works!... almost perfect - set rotation seems to still think the display is 128x160 even though I have set it to 128x128 in ILI9163_Defines.h and the TFT_eSPI constructor. When using this code:
```
for (int i = 0; i < 4; i++)
  {
    tft.setRotation(i);
    tft.fillScreen(TFT_DARKGREEN);
    tft.drawRect(0,0,128,128, TFT_YELLOW);
    tft.fillRect(2,2,10,10, TFT_RED);
    tft.fillRect(128-12,128-12,10,10, TFT_BLACK);
    tft.setCursor(12, 128/2, 1);
    tft.setTextColor(TFT_WHITE,TFT_BLUE);  tft.setTextSize(1);
    tft.println("Hello World!");
    delay(2000);
  }
```
I get this:
[image]

Bodmer, 2017-04-18:
OK, the offsets are expected.
Now I know the display driver works OK with your display. I will update the library later to correct those.

larsenglund, 2017-04-18:
Awesome, thanks again!

Bodmer, 2017-04-18:
OK, have uploaded the offset fix.

You need to define the display size in the setup.h file inside the library for this to work, e.g. in "User_Setup.h" (or in you custom user setup file in the "User_Setups" folder).

closed 2017-04-18 by Bodmer

larsenglund, 2017-04-19:
Hi, the image it is constantly cropped now instead of only being cropped at certain rotations.

Bodmer, 2017-04-19:
OK, thanks. It was a 50:50 guess which end of the display CGRAM the vendor had wired the pixels!

Bodmer, 2017-04-23:
The offsets have been corrected in the master library. Thanks.