Bodmer/TFT_eSPI #11: Question: Pin Setup Kuman 3.5 Inch

steffenmllr, 2017-04-18:
I had a display laying around [Info1](https://www.amazon.de/gp/product/B01CNLYL1C) [Info2](http://kumantech.com/kuman-35-inch-tft-lcd-display-480x320-rgb-pixels-touch-screen-monitor-for-raspberry-pi-3-2-model-b-b-a-a-module-spi-interface-with-touch-pen-sc06_p0014.html) [Info3](https://mega.nz/#F!HUFQFDwL!fhM7RPrhbMJLsZ7ISLEr0A!HRFjEbwZ)  but I'm not really sure how to wire it and which settings to use. This is what I googled for the display:

```
1、17               3.3V       ------   Power supply positive (3.3V power input)
2、4               5V         -- ----     Power supply positive (5V power input)
3、5、7、8、10、12、13、15、16    NC         --- ---         NC
6、9、14、20、25                    GND        -------          Power GND
11               TP_IRQ     ------ Touch panel interrupt, detect the touch panel is pressed down to a low level
18               LCD_RS       -------   Instruction / data register selection
19               LCD_SI / TP_SI   ------ LCD display / touch panel SPI data input
21               TP_SO          -----  SPI data output of touch panel
22               RST                -----        reset
23               LCD_SCK / TP_SCK   LCD  --------    SPI clock signal for display / touch panel
24               LC _CS           --------          LCD chip select signal, low level LCD
26               TP_CS    ---------   The touch panel chip select signal, low level selection of touch panel
```

I'm new to this hardware thing :) I also have a wemos d1 mini and some nodemcu here. What would be the minmal setup just to draw some text on the display? I did use the `RPI_ILI9486_DRIVER` and it compiled and flashed to the nodemcu but there was nothing on the display.

Bodmer, 2017-04-18:
These are the connections I use for a NodeMCU board (assumes touch display is not used).

- Display SCK to NodeMCU pin D5
- Display MOSI  to NodeMCU pin D7
- Display DC to NodeMCU pin D3
- Display RST to NodeMCU pin D4 (or RST)
- Display CS to NodeMCU pin D8
- Display 0V (GND) to NodeMCU pin GND (0V)
- Display +5V to NodeMCU 5V (Vin)
- Display 3.3V(TP_CS) to NodeMCU 3.3V

[image]

closed 2017-04-18 by Bodmer

steffenmllr, 2017-04-20:
@Bodmer Thank you! Got it working!