Bodmer/TFT_eSPI #13: Missing glcdfont.h ?

Aggebitter, 2017-04-22:
noticed that I can't get font 1 (glcdfont) working

In User_Setup.h: (as in default settings)

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4 
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

Code:
tft.drawString(firstLine , charColOffset,charRowOffset + hight /2 + 20);            // No characters !
tft.drawString(secondLine , charColOffset,charRowOffset + hight /2 + 20*, 1); // No characters !
tft.drawString(thirdLine , charColOffset,charRowOffset + hight /2 + 20*3, 2);  // Works

Bodmer, 2017-04-22:
Can you provide example code that does not work?

I tries this and it seems to work OK:

#include <SPI.h>
#include <TFT_eSPI.h> // Graphics and font library for ILI9341 driver chip
TFT_eSPI tft = TFT_eSPI();  // Invoke library

void setup(void) {
  tft.init();
}

void loop() {
  tft.fillScreen(random(0x10000));
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.drawString("Hello World!", 0,  0, 1);
  tft.drawString("Hello World!", 0, 20, 2);
  tft.drawString("Hello World!", 0, 60, 4);
  delay(10000);
}

Aggebitter, 2017-04-23:
I think it's an preprocessor issue.
I'm using it TFT_eSPI.h  in a external cpp, header and then font "1" does not work 

Will write some code example for test soon ... 

(I'm making a "OSD Keyboard" for configuring the ESP and use it as an external cpp function so that the keyboard can be called easy from whit in the the arduino sketch and just write it's keyboard buffer to an char array in the same sketch. The code is to ugly to make public yet.)

Aggebitter, 2017-04-23:
The code i'm having problems with. Probably my own fault someware
Uses default pinout for TFT_eSPI  no reset pin for TFT and CS for touch is Wemos D1 R2 mini pin "D2"
https://github.com/Aggebitter/OSD_Keyboard

Bodmer, 2017-04-23:
OK, I will close this for now. If you find the library does have a bug then I will re-open.
Thanks.

closed 2017-04-23 by Bodmer