Bodmer/TFT_eSPI #15: pgmspace error

reaper7, 2017-04-28:
Looks like latest commit on master branch esp8266/Arduino https://github.com/esp8266/Arduino/commit/0b672668bf636ef650b8b700a0af14677c3fafe5
causes errors in compilation.
I got:
```
C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual size_t TFT_eSPI::write(uint8_t)':

C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:2613:89: error: invalid conversion from 'unsigned int' to 'const void*' [-fpermissive]

       width = pgm_read_byte( pgm_read_dword( &(fontdata[textfont].widthtbl ) ) + uniCode-32 );

                                                                                         ^

In file included from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI/Fonts/glcdfont.c:6:0,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.h:36,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:20:

C:\PROGRAMY\arduino\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:102:23: error:   initializing argument 1 of 'uint8_t pgm_read_byte(const void*)' [-fpermissive]

 static inline uint8_t pgm_read_byte(const void* addr) {

                       ^

C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'int16_t TFT_eSPI::drawChar(unsigned int, int, int, int)':

C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:2761:76: error: invalid conversion from 'unsigned int' to 'const void*' [-fpermissive]

       width = pgm_read_byte( pgm_read_dword( &(fontdata[font].widthtbl ) ) + uniCode );

                                                                            ^

In file included from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI/Fonts/glcdfont.c:6:0,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.h:36,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:20:

C:\PROGRAMY\arduino\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:102:23: error:   initializing argument 1 of 'uint8_t pgm_read_byte(const void*)' [-fpermissive]

 static inline uint8_t pgm_read_byte(const void* addr) {

                       ^

C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:2786:54: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

           line = pgm_read_byte(flash_address + w * i + k);

                                                      ^

In file included from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI/Fonts/glcdfont.c:6:0,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.h:36,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:20:

C:\PROGRAMY\arduino\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:102:23: error:   initializing argument 1 of 'uint8_t pgm_read_byte(const void*)' [-fpermissive]

 static inline uint8_t pgm_read_byte(const void* addr) {

                       ^

C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:2826:54: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

           line = pgm_read_byte(flash_address + w * i + k);

                                                      ^

In file included from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI/Fonts/glcdfont.c:6:0,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.h:36,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:20:

C:\PROGRAMY\arduino\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:102:23: error:   initializing argument 1 of 'uint8_t pgm_read_byte(const void*)' [-fpermissive]

 static inline uint8_t pgm_read_byte(const void* addr) {

                       ^

C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:2870:43: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

         line = pgm_read_byte(flash_address);

                                           ^

In file included from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI/Fonts/glcdfont.c:6:0,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.h:36,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:20:

C:\PROGRAMY\arduino\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:102:23: error:   initializing argument 1 of 'uint8_t pgm_read_byte(const void*)' [-fpermissive]

 static inline uint8_t pgm_read_byte(const void* addr) {

                       ^

C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:2926:45: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'const void*' [-fpermissive]

         line = pgm_read_byte(flash_address++); // 8 bytes smaller when incrementing here

                                             ^

In file included from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI/Fonts/glcdfont.c:6:0,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.h:36,

                 from C:\PROGRAMY\arduino_projekty\libraries\TFT_eSPI\TFT_eSPI.cpp:20:

C:\PROGRAMY\arduino\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:102:23: error:   initializing argument 1 of 'uint8_t pgm_read_byte(const void*)' [-fpermissive]

 static inline uint8_t pgm_read_byte(const void* addr) {

                       ^
```

Bodmer, 2017-04-28:
I suggest you use the stock 2.3.0 board package installed via the IDE until the bugs and incompatibilities are weeded out of the latest esp8266 package.

Bodmer, 2017-05-01:
I have a closer look at the errors and it seems the byte address must be explicitly declared as a byte pointer to work with the updates esp8266 core.

I have updated the library, so all should be OK now.

closed 2017-05-01 by Bodmer

reaper7, 2017-05-02:
👍