Bodmer/TFT_eSPI #17: Compilation error.

harshcht, 2017-05-11:
Every time I try to compile the rainbow examples in eSPI library it gives an error , SPIMMOSI,SPIBUSY and many other variables are not defined. These are being used in TFT_eSPI.cpp but I cannot find where are they defined or innitialised.

Bodmer, 2017-05-20:
I have added ESP32 support.
Define the pins you use in User_Setup.h
Only tested with an ILI9341 so far.

Let me know if you have problems.

harshcht, 2017-05-20:
It is now working for esp-32 and st7735 , but red and blue color are getting exchanged.
i.e. it displays red in place of blue and vice versa.

Bodmer, 2017-05-22:
The BLACKTAB and REDTAB have swapped colours compared to GREENTAB so try all the TAB options.

Bodmer, 2017-05-22:
I've tried the ST7735 driver and it works OK. So can you check your setup and make sure the TFT really is a ST7735. As I cannot find a fault with the library I am closing this issue.

closed 2017-05-22 by Bodmer