Bodmer/TFT_eSPI #19: Question

MarcFinns, 2017-05-23:
Hi Bodmer, back again and just saw the ESP32 compatibility - great!!
A quick question. I am looking for fonts on various sizes. I like the ones you embedded in your porting of the weather station but only two sizes are available.
Is your lib compatible with Paul Stoffregen collection here on Github, and derived? (e.g. https://github.com/sumotoy/commonFonts).
thanks once more. m

Bodmer, 2017-05-23:
No the library is not compatible with Paul's fonts.

You can generate fonts in the correct format using the online utility here:
[http://oleddisplay.squix.ch/#/home](http://oleddisplay.squix.ch/#/home)

and then use them as custom fonts.

closed 2017-05-23 by Bodmer

pipi61, 2017-05-23:
Yes, use squix converter, but large font this converter not generate,
or use in https://github.com/adafruit/Adafruit-GFX-Library  ->fontconvert utility (first make executable fontconvert). This utility convert any windows ttf font.

Bodmer, 2017-06-17:
True, the Squix utility only generates fonts up to size 100 (height in pixels)..

beegee-tokyo, 2017-10-11:
Sorry to comment on a closed issue, but [http://oleddisplay.squix.ch/#/home](http://oleddisplay.squix.ch/#/home) is down. I tried to compile [https://github.com/adafruit/Adafruit-GFX-Library ->fontconvert](https://github.com/adafruit/Adafruit-GFX-Library), but doesn't get it to work.

Is anybody aware of another online or Windows tool to create fonts for this lib?

Bodmer, 2017-10-11:
The font generator website does go down sometimes. Try leaving a message on Dani's [suport forum.](https://blog.squix.org/2017/07/finally-support-forum.html)

pipi61, 2017-10-11:
try adafruit fontconvert, upload binary:
http://puzsar.hu/adafruitfontconvert.rar

beegee-tokyo, 2017-10-12:
@pipi61 thank you I will give it a try.