Bodmer/TFT_eSPI #20: Issues with setRotation()

harshcht, 2017-05-26:
the tft works fine normally but whenever I use the setRotation() function it behaves aierdly.
For instance , 

     `#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
     #include <SPI.h>

    TFT_eSPI tft = TFT_eSPI(); 

     void setup() {
    // put your setup code here, to run once:
     Serial.begin(115200);
    tft.init();
    //tft.setRotation(1);
     tft.fillScreen(ST7735_GREEN);

    Serial.println(tft.width());
    Serial.println(tft.height());
    tft.fillRect(0,0,tft.width()/4,tft.height()/4,ST7735_RED);
    }

    void loop() {
    }` 

This code works perfectly fine but if I uncomment the line
       `tft.setRotation(1)`
the display suddenly stops i.e nothing is displayed, the tft goes entirely white.

Now when I exchange red and green (keeping setRotation() uncommented) this is what is displayed : 

[image]

Bodmer, 2017-05-26:
I don't have a similar setup I can test at the moment but try adding a 20ms delay before and after the setRotation() call.  Some TFT drivers need a delay after some commands so maybe this is the problem. I will check the ST7735 datasheet.

harshcht, 2017-05-27:
Thanks @Bodmer . Adding the delays partially solves the problem, but it inverts the blue and red color again.

Bodmer, 2017-06-16:
I have tried my ST7735 display and it works fine so I am unable to help you at the moment.  can only assume that the display you have is not an ST7735. Does it work normally with the Adafruit libraries and a standard UNO?

harshcht, 2017-06-17:
I checked it on arduino mega. On that it works fine.

Bodmer, 2017-06-17:
Can you describe your hardware setup in more detail. Can you provide a web link to the display vendors page.

Bodmer, 2017-06-17:
Also try the latest version of the library.

closed 2017-06-23 by Bodmer