Bodmer/TFT_eSPI #22: Jpeg MCU compatibility with pushColors function

WestLight, 2017-06-15:
I've really appreciated your work on SD, Jpeg and decent fonts. I was originally using Adafruits ILI9341 library, and found it slow and without attractive fonts. I changed the code for a Jpeg slide show, reading an SD card, to use TFT-eSPI and ran into a problem with 16x16 MCUs. The original code for pushColors anticipated 255 max pixels which results in a missing pixel in every 16x16 MCU. I changed the unit8_t len to unit16_t len as below in both the .h and .cpp files which still left the second pushColors function unique so there is no ambiguity for the compiler. I'm not certain if it would have been better to create a new say, pushMCU, or the changes I made. Now it's working very well, reading Jpegs instead of bitmaps. Great thanks to all who contributed.

void TFT_eSPI::pushColors(uint16_t *data, uint16_t len)

Bodmer, 2017-06-16:
Hi, Thanks for suggesting this. The pushColors() functiom took an 8 bit value for legacy reasons.

If you use the JpegDec.readSwappedBytes() function instead of JpegDec.read(), then you can use the pushRect() function which sets the window for you and also pushes all the pixels to the display. You will probably find this is faster too. So you code would look something like this:

    // read each MCU block until there are no more
    while (JpegDec.readSwappedBytes()) {
	  
      // save a pointer to the image block
      pImg = JpegDec.pImage ;

      // calculate where the image block should be drawn on the screen
      int mcu_x = JpegDec.MCUx * mcu_w + xpos;  // Calculate coordinates of top left corner of current MCU
      int mcu_y = JpegDec.MCUy * mcu_h + ypos;

      // check if the image block size needs to be changed for the right and bottom edges
      if (mcu_x + mcu_w <= max_x) win_w = mcu_w;
      else win_w = min_w;
      if (mcu_y + mcu_h <= max_y) win_h = mcu_h;
      else win_h = min_h;

      // draw image MCU block only if it will fit on the screen
      if (( mcu_x + win_w ) <= tft.width() && ( mcu_y + win_h ) <= tft.height())
      {
       tft.pushRect(mcu_x, mcu_y, win_w, win_h, pImg);
      }
      else if ( (mcu_y + win_h) >= tft.height()) JpegDec.abort(); // Image has run off bottom of screen so abort decoding
    }

closed 2017-06-16 by Bodmer

WestLight, 2017-06-17:
@Bodmer
Slick! Works great! I thought this ESP32 was fast. Now writing the Jpeg MCUs to display is even faster. Thanks!