Bodmer/TFT_eSPI #23: 'class TFT_eSPI' has no member named 'getTextBounds'

Ierlandfan, 2017-06-16:
Can you implement the function?

Bodmer, 2017-06-16:
The getTextBounds() member function was added by Adafruit because their library did not support the rendering of text in the "FreeFont" format with a background.  The TFT_eSPI library does support rendering with a background so I have not had a need for that function.  There is also a function "setTextPadding()" that sets the bounding box width in pixels for the background, so that can be used to ensure small/short numeric values over-write long ones.

As the font rendering in the library does allow background it does do the maths to calculate the bounding box, so in principle I could add the function requested if those available do not suit your needs.

closed 2017-06-23 by Bodmer

tablatronix, 2020-04-01:
It would be nice to have a wrapper for porting existing code easier