Bodmer/TFT_eSPI #24: fillScreen not working

techpad-marcb, 2017-07-08:
I have everything working on my esp8266 12F unit with 1.8 Greentab tft accept fillScreen. For some reason  cannot clear the screen. It always show what was previously loaded.

Any idea what i'm doing wrong?

techpad-marcb, 2017-07-08:
Never mind everything on my screen is mirrored and does not matter which color tab i pick they are all doing the same thing.

Do i have a screen that is just not supported?

Bodmer, 2017-07-11:
Unfortunately there are multiple ways that theST7735 controller can be wired to the TFT by the screen manufacturer.  There is a control register in the display that allows the mirroring setting.

These control bits for a ST7735 are [set in this code](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ST7735_Rotation.h) that also sets the rotation and the RGB order of bits (RGB or BRG bit order).

The bits to try changing to suit your display are TFT_MAD_MX, TFT_MAD_MY and TFT_MAD_MV

These are the mirror X, mirror Y and mirror Vertical bits.

closed 2017-07-24 by Bodmer

MHz000, 2017-10-08:
Thank-you Bodmer, great work!
As  techpad-marcb commented on 8 Jul clear the screen (**tft.fillScreen(TFT_GREY)**; dosn't work. To check first run the rainbow example and then one of the clock examples.  
mirroed sreen? Simply **tft.setRotation(-3);** will do the trick

techpad-marcb, 2017-10-11:
Fantastic will keep that in mind i did find another library that worked with the screen but will keep this in my back pocket for future projects.