Bodmer/TFT_eSPI #26: SPI overlap support

fiskn, 2017-07-12:
It appears that SPI overlap support has recently been added to the arduino esp8266 code. 

https://github.com/esp8266/Arduino/pull/3189

This would be really useful for projects which are pin constrained and where the screen is only updated infrequently.

From what I can understand, due to it requiring use of the hardware CS pin, it has to use write/transferBytes functions so that all data is transmitted in one go. From looking through the code in your library,  it looks to me that single byte transmissions are being used.

Do you think that your library could be modified to use overlapping SPI, or are their specific reasons why this would not be possible?

Bodmer, 2017-07-15:
As far as I can see, with one extra line in init() the libray should work fine in overlap mode. Single byte transactions are needed in some cases due to the need to toggle the DC line. I suspect the the need to use burst SPI transactions is only a constraint imposed by the SPI RAM chip used by the author to test the new addition. I will try out the mode later today.

Bodmer, 2017-07-15:
OK. I have tested the overlap mode with a ST7735 display and it works. It just needed a couple of conditional lines in init().

I will upload the change later for you to try with your display.

Bodmer, 2017-07-15:
I have uploaded the changes. Post back if it does not work.

The "Setup9_ST7735_Overlap.h" setup has been used for testing.

closed 2017-07-15 by Bodmer

fiskn, 2017-07-17:
Can confirm overlap SPI is working for me. Awesome work!!!

Bodmer, 2017-07-17:
Thanks for letting me know, good luck with your project.