Bodmer/TFT_eSPI #27: Button class from Adafruit_GFX_Button

universam1, 2017-07-14:
Awesome work with this library!

Would it be possible to also port the Adafruit_GFX_Button https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_GFX.h#L145 as to support touch based UI functions?

Thanks!
Sam

Bodmer, 2017-07-15:
Hi, this is on my "to do" list. It may be a while before it gets done as I have not needed it in my projects...

Bodmer, 2017-07-15:
Can you point me to an example sketch that uses the buttons functions of the Adafruit_GFX library?

universam1, 2017-07-16:
Awesome, thank you very much!

Here is a nice tutorial how to use it. 

https://learn.adafruit.com/arduin-o-phone-arduino-powered-diy-cellphone/arduin-o-phone-sketch

Honestly, in meantime I found another library that contains more sophisticated UI functions written for the Adafruit. I wonder if it makes more sense to port it to your awesome driver!?
https://github.com/ginge/GuiLibrary
I could try help with this.

Bodmer, 2017-07-17:
The GuiLibrary could be adapted as a stand-alone library to work with the TFT_eSPI library but I do not have time (at the moment) to do this.

ImpulseAdventure, 2017-09-11:
FYI -- I have now added support for TFT_eSPI into the [GUIslice](https://github.com/ImpulseAdventure/GUIslice) library, so this could be another option for consideration.

I tested TFT_eSPI + GUIslice with a ST7735 OLED as well as a ILI9341 TFT.
So far, I have been very impressed with the performance of the TFT_eSPI library -- great job!

Bodmer, 2017-09-26:
Thanks for the feedback and information. I've been off traveling so have had no time to look at updates but as winter approaches now I should get more time to investigate GUIslice.

Bodmer, 2017-10-20:
The GUIslice library works well, I had to add a library.properties file to get it recognised but after updating the GUIslice_config.h file the examples all ran on an RPi display.

I can't see how to adapt the library for the SPI XPT2046 touch controller on the display (basic support for this controller is now within the TFT_eSPI library), but there are also other third party libraries that support that touch controller.

Bodmer, 2017-10-23:
I have added the Adafruit_GFX compatible Button class and keypad demo examples.

The touch screen sensing is now more reliable by using touch pressure and double sampling.

Next step is to work with GUIslice.

closed 2017-10-23 by Bodmer