Bodmer/TFT_eSPI #28: Add support for ILI9488 (Parallel)

Ierlandfan, 2017-07-16:
I saw there was a merge request for a parallel version of another display (ILI9486 I think t was)
How hard would it be to modify or add support for a ILI9488 parallel version?

Bodmer, 2017-07-17:
Yes, you can find the [parallel library here](https://github.com/konkrog/TFT_eSPI).
You would need to create the [TFT_Driver](https://github.com/Bodmer/TFT_eSPI/tree/master/TFT_Drivers) for the ILI9488 controller, the ILI9486 driver could be used as a template.  As I have no way of testing the parallel code I am unable to debug any adaptions.  Where did you get your display from, is it an 8 bit parallel interface?

Ierlandfan, 2017-07-17:
Yes it's an 8 bit parallel interface, originally made for a Arduino Mega which I also have but decided to hook it up to an ESP32. Lot of wiring, worked out ok.  I found a lot of info for the device in this forum
(https://github.com/notro/fbtft/issues/311)

Ierlandfan, 2017-07-18:
If I am not mistaken these are the steps to be taken 
Create ILI9481_init.h (Used your Arduino version of the ILI9481 for the values)
Create ILI9481.Defines.h (As test I copied those from ILI9486)
Create ILI9481.Rotation.h (Used your Arduino version of the ILI9481 for the values)
Modify User_Setup.h, User_Setup_Select.h and TFT_eSPI.h.cpp to include the new driver.
Not sure: Select ILI9481 in Setup.h 

(Please confirm, I already tried but failed)
It gives me errors, related to the font types. 

The unmodfied parallel version is not compiling, gives me errors  
"No return type given" in the readcommand section. 
I sort of get the "why" part but I am lacking the knowledge to do something about it.
Hashing them out left me with the same errors in the font types.

Ierlandfan, 2017-07-19:
Well, after seeing the same "errors" as warning when compiling with ESP8266 compiler I changed ESP32 compiler options. Now it compiles and I see something on the screen. At the moment it runs the sketch UTFT demo with a lot of artifacts and noise but hey, it compiles and I see in the noise the sketch. I can clearly without noise see the message on a red background so it could be some settings or the speed. 
Can be solved! 

"That's it, restarting in a few seconds, ESP32 Runtime is (Avg) 1245 (msecs) 8bit parallel that is.
Compared to a (SPI) ILI9486 on a ESP8266: (Avg) (3079 msec)

Ierlandfan, 2017-07-22:
Learning every day! I made the driver for the ILI9481 and it works. Learned a lot about init and rotation.
By the way, I even tried ILI9431 and that works also but the rotation, some text and the colors are off but I used that as the base and worked along the way. Even PlaneSpotter works now on the ESP32. That's another story but will fork the ESP-Parallel version and write down all. You can close this one

Bodmer, 2017-07-22:
Great, it sounds like you are making good progress. When you have done the documentation I will rig up an 8 bit display to try it out.

Ierlandfan, 2017-07-22:
I updated the documentation and made the fork available
https://github.com/Ierlandfan/TFT_eSPI

Rotation (1) works, 2 and 3 also but they need some changes, Maybe you'll able to find some other small errors since you have experience with the ILI9481.

Bodmer, 2017-07-24:
OK, looks good, hopefully I will be able to track down a compatible display from my UNO compatible screens and try it this weekend with the ESP32.

MikeA1s, 2018-02-23:
Good work! Works well. How do you go about implementing touch on esp32 with ILI9481?

Bodmer, 2018-03-01:
I have experimented with 8 bit parallel displays and the ESP32 based on changes made by Konrog and Irelandfan.

Results so far are here:
[https://github.com/Bodmer/TFT_eSPar](url)

This is an experimental version renamed and speed enhanced. It does not have the latest features such as Sprites and Smooth fonts but I will look at merging in 8 bit parallel display support in the future.

Thanks for your inputs.

Bodmer, 2018-03-10:
The main TFT_eSPI repository has now been updated to support 8 bit parallel displays with the ESP32.

closed 2018-03-10 by Bodmer

laurentopia, 2018-04-16:
Which controllers are // and which cheap screens do you folks recommend?

josephsr, 2019-02-09:
Does anyone have a schematic for ILI9481 or any display that can be interfaced with ESP32. I have got this one - https://www.aliexpress.com/item/2-4-2-8-3-2-3-5-inch-TFT-LCD-display-screen-touch-panel-40/32837085500.html?spm=a2g0s.9042311.0.0.18914c4dnnNfUn but not sure the schematics are same for the all the PCBs