Bodmer/TFT_eSPI #29: PlatformIO usage

universam1, 2017-07-20:
I am amazed your great library is available in PlatformIO! 😃 

I'm just struggling with the User_Setup that is to be done by basically modifying the library files itself, which is not the way PIO dynamically fetches libraries.
Any hint on this regard?

Thank you!

Bodmer, 2017-07-22:
Edit User_Setup to suit your hardware, or driver pins used etc, then all the example programs should run without mods. The library is designed for use with the Arduino IDE, I have not used PlatformIO.

closed 2017-07-22 by Bodmer

fiskn, 2017-08-07:
Just to comment. It would be very useful if the library by default took defines passed in via the main project source file as authoritative. Pulling down the new version of the library will override any existing settings and also makes it harder if you have multiple projects with different pin outs.

Is there something that can be done where you "#define" your user setup in your main source file  before including the TFT_eSPI header file?

universam1, 2017-08-07:
I was wondering to try to wrap your defines around a `#ifdef ...` so it would be possible to set the defines in the project but in the library. This would make the library more reusable.

Bodmer, 2017-08-11:
The Arduino IDE unfortunately does not allow configuration of the library at compile time from the calling sketch, hence the approach used with the setup files. I have not used PlatformIO so am unable to advise a way forward for you.