Bodmer/TFT_eSPI #30: Feature request (Framebuffer)

Ierlandfan, 2017-07-22:
Would it be possible to write or create a framebuffer or is there another way of doing that in this library?

Bodmer, 2017-07-24:
I'm not planning to incorporate a frame buffer as it consumes vast amounts of RAM if 16 bit colour is needed and it will need some quite drastic changes to the graphics functions.

You might like to try the [library here](https://github.com/squix78/minigrafx) if you are using an ESP8266.

closed 2017-07-24 by Bodmer

Ierlandfan, 2017-07-25:
Seems not like a big issue porting that to TFT_eSPI.  

Would you mind looking at this function and tell me how to modify this function for TFT_eSPI? It must be quite easy I guess. I can't find a similar function so I can't work it out. (I tried, it gave me garbage)
That's all that there is to it. 
```
void ILI9341_SPI::writeBuffer(uint8_t *buffer, uint8_t bitsPerPixel, uint16_t *palette) {

    if (hwSPI) spi_begin();
    setAddrWindow(0, 0, _width - 1, _height -1 );

    digitalWrite(_dc, HIGH);
    digitalWrite(_cs, LOW);

    // line buffer is in 16bit target format
    uint8_t lineBuffer[_width * 2];
    uint16_t pos;
    uint8_t bufferByte;
    uint8_t paletteEntry;
    uint16_t color;
    uint8_t shift;
    uint8_t mask =  (1 << bitsPerPixel) - 1;
    uint8_t packagesPerBytes = 8 / bitsPerPixel;
    uint16_t bytePos = 0;
    uint16_t pixelCounter = 0;
    uint16_t bufferSize = _width * _height / packagesPerBytes;
    uint8_t bytesPerLine = _width / packagesPerBytes;
    uint16_t x = 0;
    for (uint16_t y = 0; y < _height; y++) {

      for (uint16_t b = 0; b < bytesPerLine; b++) {

        for (uint8_t p = 0; p < packagesPerBytes; p++) {
          x = b * packagesPerBytes + p;
          bufferByte = buffer[bytePos];
          shift = p * bitsPerPixel;
          paletteEntry = (bufferByte >> shift) & mask;
          color = palette[paletteEntry];
          lineBuffer[x * 2] = color >> 8;
          lineBuffer[x * 2 + 1] = color;
        }
        bytePos++;
      }
      SPI.writeBytes(lineBuffer, _width * 2);
    }

    digitalWrite(_cs, HIGH);

    if (hwSPI) spi_end();
}
 
```
My modified version:
 
```
/***************************************************************************************
** Function name:           Writebuffer
** Description:             Pushes the RAM framebuffer buffer to TFT
***************************************************************************************/
void TFT_eSPI::writeBuffer(uint8_t *buffer, uint8_t bitsPerPixel, uint16_t *palette) {

  spi_begin();

// We probably need to set the addresswindow here?
  
  // line buffer is in 16bit target format
    uint8_t lineBuffer[_width * 2];
    uint16_t pos;
    uint8_t bufferByte;
    uint8_t paletteEntry;
    uint16_t color;
    uint8_t shift;
    uint8_t mask =  (1 << bitsPerPixel) - 1;
    uint8_t packagesPerBytes = 8 / bitsPerPixel;
    uint16_t bytePos = 0;
    uint16_t pixelCounter = 0;
    uint16_t bufferSize = _width * _height / packagesPerBytes;
    uint8_t bytesPerLine = _width / packagesPerBytes;
    uint16_t x = 0;
    for (uint16_t y = 0; y < _height; y++) {

      for (uint16_t b = 0; b < bytesPerLine; b++) {

        for (uint8_t p = 0; p < packagesPerBytes; p++) {
          x = b * packagesPerBytes + p;
          bufferByte = buffer[bytePos];
          shift = p * bitsPerPixel;
          paletteEntry = (bufferByte >> shift) & mask;
          color = palette[paletteEntry];
          lineBuffer[x * 2] = color >> 8;
          lineBuffer[x * 2 + 1] = color;
        }
        bytePos++;
      }
      SPI.writeBytes(lineBuffer, _width * 2);
    }

 
  CS_H;
  spi_end();
}

```

Bodmer, 2017-07-28:
Your modified code looks OK but I have not tested it.

You do need to set the address window:

setAddrWindow(0, 0, _width - 1, _height -1 );

What do you intend to write to the frame buffer?

Bodmer, 2017-11-30:
I have implemented a Sprite class that may be of interest. See library ReadMe and examples.