Bodmer/TFT_eSPI #32: Any thoughts of ESP-IDF?

vstath, 2017-08-13:
Hello,

are there any thoughts to also convert the code to natively support ESP-IDF for ESP32?

It takes a bit of effort to compile using ESP-IDF with arduino components, with some not very pretty things to do for things to compile properly. 

Thanks

Bodmer, 2017-08-13:
I have not tried the ESP IDF but maybe later this year I will have time to try it.  It looks like the IDF has a graphic library built in so with a compatible display that may be the easiest way forward for you at the moment.

closed 2017-08-13 by Bodmer

colesnicov, 2019-02-23:
Hello. Any progress?

colesnicov, 2019-02-23:
https://github.com/Bodmer/TFT_eSPI/issues/158
https://github.com/Bodmer/TFT_eSPI/issues/145

Bodmer, 2019-02-23:
No, progres. I did look briefly at the IDF environment and decided it gave me no advantages in my projects, so there are no plans to make it compatible.

mgssnr, 2021-03-28:
Hello, it's been a couple of years since this was asked, and I thought I'd ask again.  Thank you for publishing this great library for arduino users!

FWIW, and I understand you're looking at *your* projects (and I get it completely), but nowadays there is tons more applications out there for such a small form factor board with small screens, and many folks use ESP32-IDF and I wondered if you'd reconsidered.  I get that most likely more people use Arduino's IDE and not ESP32-IDF.

(Personally, I'm looking at a TTGO board and the esp-32 homekit library, because my home security system went out a long time ago and I'd like to allow my phone to see the state of various points in the house.)

I could port what you've done to ESP32-IDF since you've done the heavy lifting.  It would just take a little time to figure out what needs adjusted.  (And, of course, if someone's already ported it...)

Thanks again!

colesnicov, 2021-07-19:
@mgssnr Hello.  I have a working library for ILI9341 and touch under ESP8266-RTOS-SDK.  I modified the library from [Magore](https://github.com/magore/esp8266_ili9341).  But I still have it unfinished, so I won't share it yet, but if you know something exists ..

alotionIG, 2022-07-07:
Hi @colesnicov , I am looking for it either, 
Have you finished modification the library? 
If that is so, could you share it?

Bodmer, 2022-07-07:
See #1770

colesnicov, 2022-07-07:
@alotionIG
 Hi. Yes, I have it somehow modified. but I somehow have it scattered around the project. I used only TFT routines from Mr. @Magore's project. 
The modification is not difficult, but I have already modified it a lot, the esp8266 does not have enough IO, I use addressing registers and other IO for control, and it is all integrated even into the routines for working with TFT. It will be easier to edit Mr. Magor's version.

alotionIG, 2022-07-11:
Sorry for being late to inform you guys,

Thanks, @Bodmer  for the link, I tried it yesterday and it works for me, the library compiled,
Sure, thank you @colesnicov for the further information, probably modifying the library I'll take it as an alternative since the library put on arduino as esp-idf component is currently working.

colesnicov, 2022-07-11:
@alotionIG Hi. I was looking at the HDD and found my very first modified library from MAGORE. I'm not sure if it works, I don't have any free ESP to try (or time really) but you could try it. Just set the pin numbers in menuconfig... 

[Modified version of routines to work with the ili9341 display](https://github.com/colesnicov/magore)