Bodmer/TFT_eSPI #33: Random colored pixels on edges

hoekbrwr, 2017-09-01:
As I found out there is a bug in the Adafruit GFX library. As a result I have random colored pixels on both edges of the display. 
The matter have been reported and subsequentely corrected. See here: http://www.stm32duino.com/viewtopic.php?t=486&start=10

This means the Adafruit lib has to exchanged for a newer version!?

Bodmer, 2017-09-16:
ESP8266 is defined by the Arduino IDE automatically if it is setup to compile for the ESP8266. Similarly for the ESP32.

Random pixels at edges indicates your display has an offset, this can be seen if you draw a rectangle of full screen width and height at 0,0.

Some of the ST7735 tab options correct for this offset, however you may have a display with an offset that is different to the ones I tested. |In this case adjusting the vales in the appropriate Init file may help, [see here](https://github.com/Bodmer/TFT_eSPI/blob/53836b6f9d4953421cc6c25ad5f46688a200267b/TFT_Drivers/ST7735_Init.h#L169) for an example

Bodmer, 2017-09-26:
If you need support to solve this then post back. I am closing this for now.

closed 2017-09-26 by Bodmer