Bodmer/TFT_eSPI #34: Clarification on PIN Connections

Tokn59, 2017-09-06:
Hello, thank you for sharing this with us !

I need some clarification on the PIN Connections of my TFT with ESP8266-E12 : 

// Display SDO/MISO  to NodeMCU pin D6 (or leave disconnected if not reading TFT)
**-> My TFT does not have SD0 (Only SCK, SDA) - Which one to use ?**
// Display LED       to NodeMCU pin VIN (or 5V, see below)
// Display SCK       to NodeMCU pin D5
-> OK, connected
// Display SDI/MOSI  to NodeMCU pin D7
**-> My TFT does not have SDI (Only SCK, SDA) - Which one to use ?**
// Display DC (or AO)to NodeMCU pin D3
// Display RESET     to NodeMCU pin D4 (or RST, see below)
// Display CS        to NodeMCU pin D8 (or GND, see below)
// Display GND       to NodeMCU pin GND (0V)
// Display VCC       to NodeMCU 5V or 3.3V

Seems I am missing a PIN ?

Many thanks for your help !!

Kind Regards,

Tony

emilekm2142, 2017-09-07:
SDA is the screen's input, connect it to the output of the esp.
MOSI stands for Master Output Slave Input
MISO stands for Master Input Slave Output
Screen is a slave and esp is a master. Screen does not output anything, so you can ignore MISO.

So SDA connects to MOSI.

Bodmer, 2017-09-26:
I've been off traveling so have been slow at responding... have you solved this now?

When SDO/MISO is missing on a display and the pin is labelled SDA then that line is typically bi-directional, in this case leave MISO on the ESP8266 unconnected.

Connecting SDA to MOSI on the ESP8266 will work but as the library does not support bi-directional data yet then you will not be able to read registers/pixels back from the TFT.

Bodmer, 2017-09-27:
I'm closing this for now, but if you are still experiencing problems then let me know and I will open it again and try to help.

closed 2017-09-27 by Bodmer