Bodmer/TFT_eSPI #39: Customizing settings

flodejr, 2017-09-20:
Hi,
Thanks for this wonderful library.  I love that you have created a total seperate library rather than forking the Adafruit libraries and naming it differently.  The has been a plethora of forks of the Adafruit TFT libraries each with subtle differences and incompabilities amongst them, leading to much confusion when a project is revisited after a long time.

However, I have a slight problem with your library in the way the settings are customized.  The customizations are done via editing of the user_setup.h file which is part of the library itself.  I work with quite a number of TFT displays and also different connections on different boards.  It is not efficient to customize the library or duplicate the library for each and every project that I work on.  The settings should be stored with other functional codes of the project rather than in the setup itself.

I could have forked your library but then it will then again soon create problems like the Adafruit libraries, too many forks, too many variants of the same name.

Hope to hear from you soon.

pipi61, 2017-09-21:
Hi!
try:
copy all the tftespi directory files/folders to the project folder
remove the tftespi files/folders from the directory folder
compile it, where the include file not found error messages, modify #include <...> to "..."

Bodmer, 2017-09-26:
I've been off traveling hence my tardiness in responding...

Unfortunately I see no way of customising the library from the sketch when using the Arduino IDE (which is the only one I use at the moment). Hence the rather awkward approach used.

Things are made slightly quicker if the "User_Setup_Select.h" file is used, as then only a quick edit is needed to select the setup.

Have you tried pipi61's suggestion?

Bodmer, 2017-09-27:
I'm going to close this as the library is targeted at the Arduino environment and thus I see no way of making the options configurable from the sketch at the moment.

closed 2017-09-27 by Bodmer

NdK73, 2017-11-19:
See "issue" #54 . That does exactly what @flodejr asked.