Bodmer/TFT_eSPI #40: ILI9844 Support

motters, 2017-09-25:
Hi,

I'm trying to create a driver for the ILI9844 for your library. I want to use ILI9844 as they come in a really nice size 3.5".

I've pretty much got it working but the colours come out weird and the image is striped (see attached image). 

I'm just asking if you have any idea what it could be? I can push my config to a repo if it helps?

[image]

Thanks

Sam

Bodmer, 2017-09-26:
I've been off traveling hence my tardiness in responding...

As you say it is close to working.  The problem may be just the init sequence for the display, or the SPI clock rate is too high.  Can you post you xxxx_Defines, xxxx_Init and xxxx_Rotation header files for me to look at?

Where did you get the display? If it is cheap then I can order one and see what is happening.

motters, 2017-09-27:
I've tried slowing down the SPI clock rate with no success. I bought the display from BuyDisplays ( 
http://www.buydisplay.com/default/lcd-3-5-inch-320x480-tft-display-module-optl-touch-screen-w-breakout-board ).

I'm going to double check there is no defect on the LCD board as it can be configured into different modes (18-bit Parallel , 3-Wire Serial SPI, 4-Wire Serial SPI etc).

I've uploaded the files to a gist https://gist.github.com/motters/38a26a66020f674b6389063932048e4c, once the lcd is working i'll tidy them up and submit a merge request if you want.

Cheers for the help pal

Bodmer, 2017-09-27:
I have looked at your init file and see that the [Interface Pixel Format setting](https://gist.github.com/motters/38a26a66020f674b6389063932048e4c#file-ili9844_init-h-L61-L62) does not look right.

It looks like you have used the [init code from here](https://github.com/jaretburkett/ILI9488/blob/master/ILI9488.cpp#L430-L431) and changed the value to try and make the RGB format 16 bits.

I think the setting should be:

writecommand(0x3A);      // Interface Pixel Format 
writedata(0x55); //16 bit  

Try that and report back.

motters, 2017-09-28:
I got the init code from the manufacture, so i guess that guy did the same. Which is why i was so puzzled that i was having problems. 

The good news is that guys library does work however even if i copy and past his init code into this library i get some what of an image but gray scale and broken. The other guys library is configuring the LCD into 18 bit colour mode (0x66) which possibly could be the issue.

When i try using 16 bit colour mode (0x55) i get a load of black strip lines on the screen, see below. 

[image]
[image]

motters, 2017-09-28:
So after some reading pal it looks like the ILI9844 only works in 24 bit colour mode when using 4 wire SPI hence the colour / broken image issue. 

How easy or hard do you think it will be to factor that into your lib?

motters, 2017-09-28:
After further reading through the datasheet it does look like the problem is due to the colour mode.

Interestingly the ILI9844 should support 16 bit colour (RGB565) in 4 bit SPI mode, which is what i am using, along with 8 (RBG111) and 18 (RBG666) bit colour.

Although in 3 bit SPI mode it will only support 8 bit colour (RBG111) and 18 bit colour (RBG666) so not 16 bit which this library uses i think.

See page 119-122
http://www.buydisplay.com/download/ic/ILI9488.pdf

So if i'm correct the only way to make this lib work with the ILI9844 is to convert 16 bit colours before they are sent, to 18 bit when the driver is ILI9844.

What do you think?

Bodmer, 2017-09-28:
The number of bits in a colour is also set by the IM[2:0] configuration pins, see section 4.7.2 on page 121 of the data sheet, this means the config link must al be soldered to set the lins high and the solder links that set the pins low removed. Have you done that? Then the pixel format command:

writecommand(0x3A); // Interface Pixel Format
writedata(0x55); //16 bit

Should be used and the display will accept 16 bit colours.

Hacking the library to support 24 bit colour would not be something I wish to do simply because I have no displays that operate in this mode.

motters, 2017-09-29:
I have set IM[2:0] high as far as i am aware. I have configured the board to SPI-4 page 13 http://www.buydisplay.com/download/manual/ER-TFTM035-6_Datasheet.pdf of the breakout board datasheet

I'll contact Buy Displays to check that the shorting points are related to IM though as they don't provide circuits and get back to you.

Bodmer, 2017-10-03:
OK, unfortunately I have no more ideas on what could be the cause of your problem.

As lines only draw approximately 2/3 of the way across the screen in your [picture here](https://user-images.githubusercontent.com/2155687/30956830-5dba7c80-a430-11e7-9c9b-c4280e476fea.jpg), it implies that the display is in 24 bit mode (only getting 2 bytes per pixel instead of 3).

closed 2017-10-15 by Bodmer

ghostkcleung, 2021-05-20:
My ILI9488 cannot set to 16bit color interface
There is blank screen after execute...
Command: 0x3A
Parameter: 0x55

It can work as 18-bit only.
But for the consideration of performance.
I prefer transfer 2bytes per pixel.

Bodmer, 2021-05-20:
The screen can only be used in 18bit colour mode with a SPI interface. In parallel mode is will work with 16 bit colours.

dariogreggio, 2022-06-08:
ziofa.