Bodmer/TFT_eSPI #41: Porting to Lua

abdollahpour, 2017-09-26:
This library is beyond fast and good. Could you see any future for porting it to Lua firmware? Lua has very nice features to be honest, much more than linear Arduino programming, the only thing is missing is a very good TFT library like this. UCG is terribly slow.

Bodmer, 2017-09-27:
Thanks for the feedback. Unfortunately I know little about Lua so it would be a steep learning curve for me.

closed 2017-09-27 by Bodmer

abdollahpour, 2017-09-28:
It's easy (for you) to write a new module. Because to write a new module you use C not Lua itself. Here is the documentation:

    https://github.com/nodemcu/nodemcu-firmware/wiki/%5BDRAFT%5D-How-to-write-a-C-module

I'm in China and I'm going to make a new type of Dekit with many features using ESP8266. It has large TFT display and having this library to driving it, would be awesome.

Bodmer, 2017-09-28:
Thanks for the link, I will have a look and see what needs to be done. I have another project that is taking up my time at the moment so you will need to be patient as it may be a few weeks before I have time to look at this.