Bodmer/TFT_eSPI #44: Feedback on ESP32 and ILI9163 display

beegee-tokyo, 2017-10-10:
IDE PlatformIO
Device ESP32 dev module
Display Elecrow 1.77" 128x160 TFT display with ILI9163 controller

Works good, but I didn't check all lib functions yet. Max SPI speed is 27Mhz (most likely limited by the display itself). Faster SPI speeds produce garbage on the screen.

Encountered one conflict with OTA updates. You initialize with
`SPI.beginTransaction(SPISettings(SPI_FREQUENCY, MSBFIRST, SPI_MODE0));`
But `SPI.endTransaction()` is never called, which causes OTA updates to fail. Don't know why.

Workaround:
In the user_setup_xxxx.h (whichever is used, I created my own for this hardware combination) uncomment `// #define SUPPORT_TRANSACTIONS` and the problem is gone.

Another issue is that I had to define TFT_MOSI, TFT_MISO and TFT_SCLK to get the communication with the display working.

Hope this feedback helps.

Bodmer, 2017-10-11:
Thanks for the feedback.
Yes the SPI speed tends to be limited by the display controller, according to the data sheet the maximum clock rate is 15.15MHz (66ns clock cycle time).
The OTA conflict is interesting, you did well to track that one down. I have not used OTA updates with the library but that is something for me to try. Thanks for the workaround info.

Some of the setup files are not updated for the ESP32 for the pin definitions (I will correct this), the top level User_Setup.h file is up-to-date and thus can be used as a reference for the pins that need to be defined. the SPI pins can be mapped to any I/O on the ESP32 so that is why they must be defined, [see here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L92-L101).

Thanks again for your feedback.

Bodmer, 2017-10-15:
I see there is a [related pull request](https://github.com/Bodmer/TFT_eSPI/pull/45) which I will track.

Bodmer, 2017-10-20:
This should be fixed now as SUPPORT_TRANSACTIONS is always defined if a ESP32 is used.

Thanks for reporting the issue and solution.

closed 2017-10-20 by Bodmer