Bodmer/TFT_eSPI #46: Usage Question: Wiring and #defines (no issue)

Gerry33, 2017-10-11:
Hi '@bodmer', 
and thanks for this nice lib ! Highly appreciated ! 

I'm confused with the wiring instructions you outlined in  'user-setup.h'.
> We must use hardware SPI, a minimum of 3 GPIO pins is needed.

Do I need only these 3 GPIOs for display  (I do not use touch capabilities) ?

If true,  are these 3 ones? :
> define TFT_CS   PIN_D8  // Chip select control pin D8
> #define TFT_DC   PIN_D3  // Data Command control pin
> #define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)

OR is the full wiring required as said in section 1  of 'user-setup.h' PLUS the 3 above ?

> // Typical setup for NodeMCU ESP-12 is :
> ...
> // Display SDO/MISO  to NodeMCU pin D6 (or leave disconnected if not reading TFT)
> ...

I do not use NodeMCU, just plain 'C'/C++' programming, but that should not make a difference.

Thanks for any help
Gerry

Bodmer, 2017-10-13:
An example setup is:
// We must use hardware SPI, a minimum of 3 GPIO pins is needed.
// Typical setup for NodeMCU ESP-12 is :
//
// Display SDO/MISO  to NodeMCU pin D6 (or leave disconnected if not reading TFT)
// Display LED       to NodeMCU pin VIN (or 5V, see below)
// Display SCK       to NodeMCU pin D5
// Display SDI/MOSI  to NodeMCU pin D7
// Display DC (RS/AO)to NodeMCU pin D3
// Display RESET     to NodeMCU pin D4 (or RST)
// Display CS        to NodeMCU pin D8 (or GND)
// Display GND       to NodeMCU pin GND (0V)
// Display VCC       to NodeMCU 5V or 3.3V

The minimum number of dedicated GPIO pins is SCK, MOSI and DC, the other display pins can be connected as noted in brackets. Some displays need the TFT CS line to be toggled during initialisation so in that caser a GPIO pin must be allocated to CS.

Gerry33, 2017-10-13:
Thank you very much for the reply  and the clarification ! 

I'm indeed struggling with  a non-working display. Its a ILI9341- display, but obviously somehow different.  Using this one [goo.gl/59SBtn](url). The pin layout is somewhat different from the well known red PCBs. I made the full wiring as you described in your *.h file, but the display does not react on anything.
From the specs of the display:

> -  Display driver chip SSD1289 / ILI9341, touch screen driver chip for XPT2046; 
> - Display Interface for 16bit parallel interface, support for 6800 and 8080 bus interface, support for the STM32 FSMC bus interface refresh speed; 
> - Touch screen interface for the SPI interface, the data line low, driving simple;  

The 2nd sentence make me nervous, 16bit parallel interface, nothing about SPI.

>  Some displays need the TFT CS line to be toggled during initialisation 

CS is wired, will try that in setup() . 
This display has 3 CS lines:  T_CS, F_CS , CS.  Wired the last CS.

Then I started your example reading the ILI- registers (TFT_Read_Reg.ino) and the display responds. 
I will now try to interpret the register responses by analysing the  specs. of the chip in the hope to get something further.

Thanks again for your kind assistance 
Gerry

Gerry33, 2017-10-13:
Edit: Likely found the solution after lots of go*ling : The display interface is a parallel one. There are some pins (IM 0..) to be closed/opened to activate SPI. No idea where these pins are. somebody obviously has solved this in the above thread at  http://www.edaboard.com/thread360074.html. 

Will try to get more information there. 
Thanks anyhow ! 
Gerry

Bodmer, 2017-10-15:
As far as I can see the display you have does not provide access to the IM[0:2] mode select pins, in that case the display will only operate in parallel mode.  Also as the display was sold as "SSD1289 / ILI9341", you may have a display that is fitted with an SSD1289 which may not be compatible with the TFT_eSPI drivers.

closed 2017-10-15 by Bodmer