Bodmer/TFT_eSPI #50: Lost user settings on library update - not an issue, just how do others save their user settings

beegee-tokyo, 2017-11-02:
I had defined my settings for a ILI9163 display connected to an ESP32 in the libraries User_Setups folder and then added it in **`User_Setup_Select.h`**
Using PlatformIO, the library was updated to the latest version today and (of course) the file I created with my display and connection settings was deleted in the process. So I needed to create it again (stupid me didn't have a backup of the file).

Just asking other users if they found a solution to keep their user settings file secure over an update of the library.

I tried to keep my settings file in the project folder and include it just before the **`TFT_eSPI.h`,** and then don't include any standard setting in **`User_Setup_Select.h`**, but that doesn't work.

Bodmer, 2017-11-02:
One approach is to create a backup copy of the example setup with your changes with a new file name (perhaps just add My to the start of the name = MySetup...) as the existing examples can easily get overwritten when updating. Of course you will still have to make a small edit to call up your particular setup file.

Unfortunately the Arduino IDE that I use does not allow library level configuration from the sketch, hence the rather inelegant method used for tailoring the library to your needs :-(

beegee-tokyo, 2017-11-03:
Ok, thanks for the feedback. I created my settings.h in my project folder and will copy it to the library folder after an update. I am using PlatformIO and it looks like PIO deletes the folder and replaces it with the new version. My settings had already a unique name and still got deleted.

closed 2017-11-03 by beegee-tokyo

Bodmer, 2017-12-17:
A change has been made to User_Setup_Select.h as a recent pull request. This will allow your sketch in PlatformIO to define the settings.

Include:

#define USER_SETUP_LOADED

and define the settings.

beegee-tokyo, 2017-12-17:
Sounds good. I'll test tomorrow and give u feedback.

Bodmer, 2017-12-17:
Info from the originator of the pull request [here](https://github.com/Bodmer/TFT_eSPI/pull/67).

beegee-tokyo, 2017-12-18:
My feedback [here](https://github.com/Bodmer/TFT_eSPI/pull/67)