Bodmer/TFT_eSPI #51: Touch part of module tjctm24024 is not working

edautz, 2017-11-03:
Hello,

I am running a tjctm24024:

http://play.fallows.ca/wp/projects/electronics-projects/tjctm24024-module-running/

The display is running fine, but the touch part is not working, whatever I try.

I run it on an ESP Doit Devboard 1.

The display part on a seperate SPI channel then the touch part:

setup.h:

// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP32 SETUP   ######

// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins

#define TFT_MISO 27
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS    22  // Chip select control pin
#define TFT_DC    21  // Data Command control pin
#define TFT_RST   17  // Reset pin (could connect to RST pin)
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TOUCH_CS 4     // Chip select pin (T_CS) of touch screen

T_DIN on D13,
T_CLK on D14
T_DO on D12

Using Arduino IDE 1.6.12
Latest arduino esp32 core version: arduino-esp32-master_v3.0-dev-1024-g92748142_24102017.zip

Can you help me out?

Bodmer, 2017-11-03:
The library assumes the touch controller and the display are on the same SPI pins, so MISO, MOSI and SCLK must be common, but the chip selects must be on separate pins.

So wire:
T_DIN to 23 (MOSI)
T_CLK to pin 18 (SCLK)
T_DO to 27 (MISO)

Let me know if this works.

edautz, 2017-11-04:
Hallo Bodmer,

Thnx for your fast respons.

I changes the wiring according your respons.

Screen remains working. Touch doesn't work either.

I modified your on_off_button_touch sketch for testing:

// Example of drawing a graphical "switch" and using
// the touch screen to change it's state.

// This sketch does not use the libraries button drawing
// and handling functions.

// Based on Adafruit_GFX library onoffbutton example.

// Touch handling for XPT2046 based screens is handled by
// the TFT_eSPI library.

// Calibration data is stored in SPIFFS so we need to include it
//#include "FS.h"

#include <SPI.h>

#include <TFT_eSPI.h> // Hardware-specific library

TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

boolean SwitchOn = false;

// Comment out to stop drawing black spots
#define BLACK_SPOT

// Switch position and size
#define FRAME_X 100
#define FRAME_Y 64
#define FRAME_W 120
#define FRAME_H 50

// Red zone size
#define REDBUTTON_X FRAME_X
#define REDBUTTON_Y FRAME_Y
#define REDBUTTON_W (FRAME_W/2)
#define REDBUTTON_H FRAME_H

// Green zone size
#define GREENBUTTON_X (REDBUTTON_X + REDBUTTON_W)
#define GREENBUTTON_Y FRAME_Y
#define GREENBUTTON_W (FRAME_W/2)
#define GREENBUTTON_H FRAME_H

//------------------------------------------------------------------------------------------
//------------------------------------------------------------------------------------------
void setup(void)
{
  Serial.begin(115200);
  tft.init();

  // Set the rotation before we calibrate
  tft.setRotation(1);

  // clear screen
  tft.fillScreen(TFT_BLUE);

  // Draw button (this example does not use library Button class)
  redBtn();
}
//------------------------------------------------------------------------------------------
//------------------------------------------------------------------------------------------
void loop()
{
  uint16_t x, y;

  // See if there's any touch data for us

  Serial.println("Enter touch");
  if (tft.getTouch(&x, &y)) {
    Serial.print("x:");
    Serial.print(x);
    Serial.print("  y:");
    Serial.println(y);

    // Draw a block spot to show where touch was calculated to be
#ifdef BLACK_SPOT
    tft.fillCircle(x, y, 2, TFT_BLACK);
#endif

    if (SwitchOn)
    {
      if ((x > REDBUTTON_X) && (x < (REDBUTTON_X + REDBUTTON_W))) {
        if ((y > REDBUTTON_Y) && (y <= (REDBUTTON_Y + REDBUTTON_H))) {
          Serial.println("Red btn hit");
          redBtn();
        }
      }
    }
    else //Record is off (SwitchOn == false)
    {
      if ((x > GREENBUTTON_X) && (x < (GREENBUTTON_X + GREENBUTTON_W))) {
        if ((y > GREENBUTTON_Y) && (y <= (GREENBUTTON_Y + GREENBUTTON_H))) {
          Serial.println("Green btn hit");
          greenBtn();
        }
      }
    }

    Serial.println(SwitchOn);

  }
}
//------------------------------------------------------------------------------------------

void drawFrame()
{
  tft.drawRect(FRAME_X, FRAME_Y, FRAME_W, FRAME_H, TFT_BLACK);
}

// Draw a red button
void redBtn()
{
  tft.fillRect(REDBUTTON_X, REDBUTTON_Y, REDBUTTON_W, REDBUTTON_H, TFT_RED);
  tft.fillRect(GREENBUTTON_X, GREENBUTTON_Y, GREENBUTTON_W, GREENBUTTON_H, TFT_DARKGREY);
  drawFrame();
  tft.setTextColor(TFT_WHITE);
  tft.setTextSize(2);
  tft.setTextDatum(MC_DATUM);
  tft.drawString("ON", GREENBUTTON_X + (GREENBUTTON_W / 2), GREENBUTTON_Y + (GREENBUTTON_H / 2));
  SwitchOn = false;
}

// Draw a green button
void greenBtn()
{
  tft.fillRect(GREENBUTTON_X, GREENBUTTON_Y, GREENBUTTON_W, GREENBUTTON_H, TFT_GREEN);
  tft.fillRect(REDBUTTON_X, REDBUTTON_Y, REDBUTTON_W, REDBUTTON_H, TFT_DARKGREY);
  drawFrame();
  tft.setTextColor(TFT_WHITE);
  tft.setTextSize(2);
  tft.setTextDatum(MC_DATUM);
  tft.drawString("OFF", REDBUTTON_X + (REDBUTTON_W / 2) + 1, REDBUTTON_Y + (REDBUTTON_H / 2));
  SwitchOn = true;
}

Serial monitor output:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:10212
entry 0x40078a00
Enter touch

What happens:

void setup() executes fine. Screen displays backgroud and button.
But when executing the void loop();
After calling  if (tft.getTouch(&x, &y)) the loop hangs. Infinite loop? 
ESP is not resetting, no watchdog error. Nothing just hangs.

Is there a way to debug this?

edautz, 2017-11-04:
Tried also two different screens, also a tjctm24028-SPI v1.1

Also Aduino Core debug level  true and Serial.setDebugOutput(true); in setup portion of the sketch.

Tried:
#define SPI_FREQUENCY   1000000
#define SPI_TOUCH_FREQUENCY  1000000

But notting matters. No change and no extra information.

Bodmer, 2017-11-04:
Mmmmm. I have only tested the touch function with an ESP8266. I have a display like yours so I will try it with an ESP32 and see what is happening. It will be a few days before this is done.

edautz, 2017-11-04:
Ok. Thanx.

Very curious.

Bodmer, 2017-11-04:
I have updated the library to work with the ESP32. It seems the switching of the SPI frequency upset the ESP32.

I have tested with an ESP32 and RPi TFT screen with the pins defined in Setup11 header file and it works. 

The board package I have loaded does not support SPIFFS with the ESP32 so I used the "Touch_calibrate" example in the Generic examples folder since that does not store the screen calibration data in SPIFFS, instead it sends the code required to the Serial port for cutting and pasting into your sketch.

Let me know how you get on with these changes.

edautz, 2017-11-05:
I tested the updated library code with  my current setup.

Works like a charm!!!! Thank you for your swift action.

closed 2017-11-05 by edautz