Bodmer/TFT_eSPI #52: Question? Using this library from different ESP32 tasks

edautz, 2017-11-09:
When using this library from different ESP32 tasks when using the RTOS tasks, the library appears to blocking and scrambling the screen output, probably due the fact that those tasks are using functions call to this library in parellel instead of in serial.

Is it possible to modify the library so this conflict can be resoved, by inplementing some kind of queing mechanism?

Bodmer, 2017-11-13:
The library is not designed to run under a real time operating system.

Can you point me to an SPI based library that does work with the RTOS, so I can see what changes might be needed?

edautz, 2017-11-14:
Frankly, I don't know.

Maybe: https://github.com/MartyMacGyver/ESP32_Adafruit_ILI9341??

I didn't test that library because the touch part wasn't described and propably missing.

I am using RTOS with your library, but as a work-a-round, only one tasks is using function calls to your library. That is working fine and stable.

Bodmer, 2017-11-16:
OK, great to hear you have a workaround. Thanks for posting this.

closed 2017-11-16 by Bodmer

rlerdorf, 2017-11-30:
This isn't really an SPI thing, is it? It is more about keeping the global properties like the cursor position from being corrupted when multiple tasks are all updating it at the same time. That can mostly be solved by having a separate TFT object for each task. Here is a 3-task ESP32 example:

```

Bodmer, 2017-11-30:
I think the issue is to due with grabbing the HAL mutex to improve graphics performance. You could try commenting out the line here and ensuring SUPPORT_TRANSACTIONS  is not defined anywhere in your setup file.

Post back if this works, I am unable to test this.

rlerdorf, 2017-12-01:
Uncommenting that #define made my example break completely. The output from task1 got completely messed up.

Bodmer, 2017-12-01:
OK!