Bodmer/TFT_eSPI #54: Simplify use of different displays in different projects

NdK73, 2017-11-18:
Currently the lib is thought for a single kind of display.
If you add include guards in User_Setup.h it becomes easy to add a file to the project, to be included before including TFT_eSPI.h .
The change would be minimal:
    #ifndef _TFT_ESPI_USER_SETUP_H
    #define _TFT_ESPI_USER_SETUP_H
    ... rest of User_Setup.h
    #endif

This way, every project can have its own setup (maybe one uses ESP8266 and another uses ESP32...).

PS: tks for your effort! This lib rocks!

Bodmer, 2017-11-19:
Thanks for your idea, I like it!

At the moment I am adding 8 bit color depth handling to the Sprite class so will finish that and then add your suggested changes.

NdK73, 2017-11-19:
It's a pleasure to be able to contribute to such a great project!

Bodmer, 2017-11-21:
I assume this will not work because sketch referenced header files are not linked into libraries,  as noted in issue #56.

I will close this, thanks for your input. Hope your project is a success.

closed 2017-11-21 by Bodmer

NdK73, 2017-11-21:
Might not work now, but it's good practice anyway.
And it's possible to make it work by editing platform.txt (see arduino/Arduino issue 5390).

Bodmer, 2017-12-17:
A change has been made to User_Setup_Select.h as a recent pull request. This will allow your sketch in PlatformIO to define the settings.