Bodmer/TFT_eSPI #55: Mirrored display

NdK73, 2017-11-19:
I'm using ST7735_DRIVER with ST7735_REDTAB variant (why not a single #define DRIVER ST7735_REDTAB ? -- but that could be for another time) with an ESP8266 (Wemos D1 mini, actually).
After pulling latest version, I get a left-right mirrored display. In a previous library version (a quite old one, but I don't remember commit ID) it worked OK. Did I miss some change? Is there a simple call to fix it?
My init code is:
TFT_eSPI disp = TFT_eSPI();
...
void setup() {
...
  disp.init();
  disp.setRotation(3);
  disp.fillScreen(TFT_BLACK);
  disp.setTextColor(TFT_WHITE, TFT_BLACK);
  disp.setCursor(12,5);
  disp.drawCentreString(title, 80, 7, 2);
}

Tks.

NdK73, 2017-11-19:
Uhm... I've found that I have to use
  disp.setRotation(5);
instead of the old
  disp.setRotation(3);
So it seems fixed.

JamesNewton, 2018-01-16:
Having the same issue with a green tab unit:
https://www.adafruit.com/product/2088
I've tried every one of the options:
````
//#define ST7735_INITB
//#define ST7735_GREENTAB
//#define ST7735_GREENTAB2
//#define ST7735_GREENTAB3
//#define ST7735_REDTAB
#define ST7735_BLACKTAB
````
and it doesn't make a difference. When `.setRotation(2)` (or 0-3) the display is always mirrored left/right. With 4 or 5 (haven't tried 6 or 7) it is correct. 

I haven't really looked into the source, but this doesn't seem like the sort of thing that should be controlled by `.setRotation. ` Should a define be added in the config to add +4 to the rotation just for ease of use?

Bodmer, 2018-01-16:
The link is for a 128x128 display. Try:
#define INITR_GREENTAB128

as this is for 128x128 displays and includes the 32 pixel offset.

The rotation and mirror control bits are set in the ST7735_Rotation.h file in the TFT_Drivers folder. Bits TFT_MAD_MV and TFT_MAD_MY control the mirroring in different axis, also the colour order RGB or BGR.
 If you change them here you must recompile and call setRotation(x) in the sketch to invoke those new setting.
Unfortunately there is no standard way the display pixels get mapped to the CGRAM and there are a lot of permutations!

Let me know what settings works in each rotation if you need to change them.

JamesNewton, 2018-01-17:
Did you mean `#define ST7735_GREENTAB128` ?
Neither make any difference for `.setRotation(2)`, it is always mirrored. Or for 0. or 1

The ST7735_GREENTAB128 option wasn't listed in the Setup9_ST7735_Overlap.h file which I'm using. I do see it now in the User_Setup.h file. Actually, that option seems to make my display only clear the left half of the display when using the `tft.fillScreen(TFT_GREY);` command (with  `.setRotation(5)`) but I can print on the entire display.

I wanted a landscape setup, so `.setRotation(5)` is working, but now that I've looked at ST7735_Rotation.h that really confuses me... rotation is mod 4 so a value of 5 should be the same as 1. In fact, the cases for 4-6 are commented out! And yet  `.setRotation(1)` is mirrored, and  `.setRotation(5)` is not!

I've only modified 2 files: User_Setup_Select.h (to uncomment:
`#include <User_Setups/Setup9_ST7735_Overlap.h>`
so that the pre-made file for overlap mode would be selected) and then  Setup9_ST7735_Overlap.h in which I set:
````
#define ST7735_DRIVER
#define TFT_WIDTH  128
#define TFT_HEIGHT 128
#define ST7735_GREENTAB128
````
And I KNOW I've modified those files because if I change values, when I recompile the sketch it changes the behavior. e.g. if I put "asldfj" into the User_Setups/Setup9_ST7735_Overlap.h file, compilation fails. 

Then I'm using the examples\160 x 128\TFT_Print_Test\TFT_Print_Test.ino as provided or with  `.setRotation(5)`

Oh... hang on... I had failed to comment out 
`#include <User_Setup.h>           // Default setup is root library folder`
in User_Setup_Select.h... wow... I don't even know how it was working at all. Good grief!

Now,  `.setRotation(1)` shows the expected display, but it's starting part way from the left. *sigh* Ok... starting over. 

It looks like
`#define ST7735_GREENTAB3` 
works correctly, except that the DEADBEEF thing wraps the F back to the next line, which makes the last line wrap back to the top. But I think that may be normal... it doesn't look like there is room for that "F" at the end of the line.

Bodmer, 2018-01-17:
ST7735_GREENTAB128 may be better for other rotations, unfortunately as there are so many ways the manufacturers wire up the pixels it is a case of try it and see.

The F will wrap as the display does not have the pixel count (ie width) needed.

JamesNewton, 2018-01-17:
Thanks again for your excellent support! Honestly, I would have given up on this long ago if you hadn't been so responsive. Let me know if I can do anything to repay your effort.

Bodmer, 2018-01-17:
Thanks for editing the posts. Good luck with your project.

closed 2018-01-17 by Bodmer