Bodmer/TFT_eSPI #56: SD7735 uses wrong colors

NdK73, 2017-11-19:
Hello again.
Seems I'm still missing something.
I tried all sub-options for SD7735_DRIVER, but it seems none sets the correct color order.
TFT_RED is displayed as blue, while TFT_BLUE is displayed as red.
At least bit order inside the color int seems OK (so 0b1100000000000000 is still blue, but not much darker than the blue displayed by TFT_RED).
I could find possibly-relevant references to BGR only in SD7735_rotation.h, but even if I change rotation (tried 0, 1, 5) the result is the same.
Moreover, using disp.fillScreen(TFT_BLUE); fills red just about half of the screen -- the other half remains black (well, actually it keeps old values).

NdK73, 2017-11-19:
A strange thing: if I use
  disp.setRotation(5);
  disp.fillScreen(TFT_RED);
only less than the left half of the display is filled with blue. But if I use
  disp.fillScreen(TFT_RED);
  disp.setRotation(5);
(so first I fill, then rotate) **all** the screen becomes blue.

Bodmer, 2017-11-20:
Different display manufacturers build their displays in different ways and one difference is the colour order (RGB or BGR). There is a bit in the "MADCTL" control register (at address 0x36)  that swaps the red and blue over. See page 110 of the [ST7735 datasheet](https://www.google.co.uk/url?sa=t&rct=j&q=&esrc=s&source=web&cd=3&ved=0ahUKEwiVo8n3iczXAhVLPRoKHY7YAqoQFgg5MAI&url=http%3A%2F%2Fwww.displayfuture.com%2FDisplay%2Fdatasheet%2Fcontroller%2FST7735.pdf&usg=AOvVaw3c7tGIh_iby2ALaWludqqW).

NdK73, 2017-11-20:
Yup. I've found TFT_MAD_BGR #define in ST7735_Defines.h, but I don't understand what I should change.
IIUC, the difference between BLACKTAB and GREENTAB/INITB is the MAD_BGR vs MAD RGB bit. But changing the 'tab' seems to have no effect...

Bodmer, 2017-11-20:
Changing the "tab" should have an effect. This implies that the wrong file is being edited.
The "User_Setup_Select.h" file by default calls up "User_Setup.h", so if you are using another setup file then it must point to it. If you are calling a setup.h file from an Arduino IDE sketch then you may find that the library uses defaults because it does not link in any header files referenced from sketches.

NdK73, 2017-11-20:
You're right. It seems Arduino IDE compiles the library independently from the sketch. :(
Enabling detailed output for compile phase gives the complete commands used. That don't include -Isketchdir ...GRRR...
Added issue 6952 ("Support in-sketch library config") to arduino/Arduino repo.
Tks for the help.

Bodmer, 2017-11-21:
OK, thanks for getting back. I will close this.

closed 2017-11-21 by Bodmer