Bodmer/TFT_eSPI #58: Crash on printing undefined chars

rlerdorf, 2017-11-28:
I was trying to print a degree symbol and was surprised to find that it caused a crash in anything other than font 1.
Reproduce sketch:

```
#include <SPI.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup() {
  Serial.begin(115200);
  tft.init();
  tft.setCursor(10, 10, 2);
  tft.setTextColor(TFT_WHITE,TFT_BLACK);
}

void loop() {
  tft.print(char(223));
  delay(100000);
}
```

The result is:
```
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400d1c4f  PS      : 0x00060330  A0      : 0x800d0b69  A1      : 0x3ffceda0  
A2      : 0x00000000  A3      : 0x00000090  A4      : 0x00000090  A5      : 0x00000084  
A6      : 0x00000010  A7      : 0x3ffc226c  A8      : 0x800d1c3c  A9      : 0x00000084  
A10     : 0x3ff44000  A11     : 0x3ffc22f0  A12     : 0x00000000  A13     : 0x00000069  
A14     : 0x00000019  A15     : 0x00000004  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000084  LBEG    : 0x400d26f0  LEND    : 0x400d2752  LCOUNT  : 0x00000002  

Backtrace: 0x400d1c4f:0x3ffceda0 0x400d0b66:0x3ffcee10 0x400e37c5:0x3ffcee40 0x400d08b1:0x3ffcee60 0x400de3c5:0x3ffcee80
```

Bodmer, 2017-11-28:
Hi, Thanks for reporting this.

The error occurs for 2 reasons:
1. The character does not exist in the font
2. Character range checking is not performed.

I will add range checking.

rlerdorf, 2017-11-28:
Yup, the first part makes perfect sense, and I didn't actually expect it to work. It was the crash that was unexpected.

MarkyAD, 2017-11-28:
Maybe textWidth() could benefit from similar range checking. I had a GFX font that contained only numbers and a decimal point, then was doing textWidth on a string that contained spaces... oops. This caused textWidth to return numbers larger than the screen width for strings only a few characters long. Using that to determine sprite size and placement led to them showing up in unintended places or offscreen.

Bodmer, 2017-11-29:
Repository updated, character code range now checked for validity.

Please raise another issue for the textWidth() problem experiences and provide an example that fails.

closed 2017-11-29 by Bodmer