Bodmer/TFT_eSPI #59: Request to include ST7789V TFT for Esp32-Wrover-kit v3

Les-A, 2017-11-29:
I would do this myself if I was capable but although there is now a library that supports the above for the basic C toolchain (loboris/ESP32_TFT_library), there does not seem to be anything available for the Arduino IDE.  Is this something you would consider including here?

Bodmer, 2017-11-29:
I do not have an ST7789V based display and this would make library modification error prone as I would not be able to test changes.  If cheap displays based on that controller become commonly available then I may buy one and have a go at supporting it, but I have too many other projects ongoing at the moment!

closed 2017-11-29 by Bodmer

Les-A, 2017-12-04:
Thanks for considering.  I imagine there are quite a lot of  these on the
market by now but I'm not sure what they are replacing.  However, the
boards do seem to have jumped in price recently.

Les-A, 2018-11-02:
Thank you so much.  I've just updated the TFT_eSPI library on my Arduino IDE and I see it now has this TFT display included.  I've tried it out on an ESP32-WROVER-KIT but although the screen flashes occasionally, nothing is displayed.  If I then press both the 'boot' and 'en' buttons together the screen lights and a clock is displayed although in the wrong colours.  As soon as I release a button the screen goes dark again.  I'll try playing around with it and if I get it to work I'll let you know what I've done.

Bodmer, 2018-11-02:
Hi, this implies the driver is not initialised correctly, check the driver is being selected correctly in the setup file.

What is happening is that the screen graphics RAM is updated by the sketch but you cannot see it. When you then press boot the screen initialisation registers are cleared to a default and then the image in the graphics RAM that was put there earlier becomes visible.

I thought that WROVER boards had a ILI9341 controller but maybe later boards have a ST7789 fitted See issue #221