Bodmer/TFT_eSPI #60: Usage Question: How to avoid stray pixels from  tft.drawString()

Gerry33, 2017-11-30:
How  can I avoid such stray pixels from tft.drawString(...);

[image]

Text datum is set to  ` 'setTextDatum (TL_DATUM)',`   
    `tft.setTextPadding(tft.textWidth(("00:00"))` is always set to the maximum widh possible.

Bodmer, 2017-11-30:
Hi, can you provide a more complete code example that illustrates the problem, including the font used. Thanks.

Gerry33, 2017-11-30:
Hi Bodmer,
thank you for taking care.

Here is a sketch that demonstrates:
```
#include <SPI.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();
void setup()
{
	Serial.begin(115200);
	delay (10);
	Serial.printf("\nStarted:" __FILE__ ", ESP-Core:<%s>,\n ESP-SDK:<%s>\n", ESP.getCoreVersion().c_str(), ESP.getSdkVersion());
	tft.begin();
	tft.setRotation(2);
	tft.fillScreen		(TFT_BLACK);
	tft.setTextColor	(TFT_YELLOW, TFT_BLACK);
	tft.setFreeFont		(&FreeSerifBold24pt7b); // this font comes with the lib
	tft.setTextDatum	(TL_DATUM);
	tft.setTextPadding(tft.textWidth("00:00"));
}
void loop()
{
		tft.drawString("00:00", 10, 10);
		delay (3000);
		tft.drawString("12:34", 10, 10);	// stray pixels
		delay (3000);
}
```
The '12:34' String shows like this
[image]

The left stray pixels are remainers of the '00:00' - string.

Bodmer, 2017-11-30:
I have fixed this is the latest version 0.17.15 in the repository.

Thanks for pointing out this issue.

closed 2017-11-30 by Bodmer

Gerry33, 2017-12-01:
Perfect ! 

> Thanks for pointing out this issue

I'm the one to give a big thank to you !   

Gerry