Bodmer/TFT_eSPI #62: [Enhancement] Shift a sprite left or right

beegee-tokyo, 2017-12-06:
I am working on a function to display temperature values as a graph on the display. And I wanted to shift(scroll???) the graph when the screen is filled and add the new data at the end. 
My display (ILI9163) has no function to shift the screen content or read back the content of the screen.

I decided to work with a sprite, but the library has neither a function to read back the sprite content for changing nor a shift(scroll???) function.

To achieve what I want, I added 2 sprite functions to the library, **`shiftRectLeft()`** and **`shiftRectRight()`**
Maybe you want to add these to the library.
Tested only with an [ILI9163 128x128 TFT](https://www.elecrow.com/144-128x-128-tft-lcd-with-spi-interface-p-855.html)  display and an ESP32.

The declarations:
```cpp
           // Shift the sprite <shift> pixels to the left and fill right side with <color>
  void     shiftRectLeft(uint32_t x, uint32_t y, uint32_t w, uint32_t h, uint32_t shift, uint32_t color);
           // Shift the sprite <shift> pixels to the right and fill left side with <color>
  void     shiftRectRight(uint32_t x, uint32_t y, uint32_t w, uint32_t h, uint32_t shift, uint32_t color);
```
The functions:
```cpp
/***************************************************************************************
** Function name:           shiftRectLeft
** Description:             shifts sprite left and fill right side with given color
***************************************************************************************/
void  TFT_eSprite::shiftRectLeft(uint32_t x, uint32_t y, uint32_t w, uint32_t h, uint32_t shift, uint32_t color)
{
  if ((x > _iwidth) || (y > _iheight) || (w == 0) || (h == 0) || !_created) return;

  for (uint32_t yp = y; yp < y + h; yp++)
  {
    for (uint32_t xp = x + shift; xp < x + w; xp++)
    {
      _img[xp-shift + yp * _iwidth] = _img[xp + yp * _iwidth];
    }
    for (uint32_t xp = x + w; xp < x + w + shift; xp++)
    {
      if (_bpp16)
      {
          _img[xp-shift + yp * _iwidth] = (uint16_t) (color >> 8) | (color << 8);
      }
      else
      {
        _img[xp-shift + yp * _iwidth] = (uint8_t)((color & 0xE000)>>8 | (color & 0x0700)>>6 | (color & 0x0018)>>3);
      }
    }
  }
}

/***************************************************************************************
** Function name:           shiftRectRight
** Description:             shifts sprite right and fill left side with given color
***************************************************************************************/
void  TFT_eSprite::shiftRectRight(uint32_t x, uint32_t y, uint32_t w, uint32_t h, uint32_t shift, uint32_t color)
{
  if ((x > _iwidth) || (y > _iheight) || (w == 0) || (h == 0) || !_created) return;

  for (uint32_t yp = y; yp < y + h; yp++)
  {
    for (uint32_t xp = x + w - 1; xp != x; xp--)
    {
      _img[xp + yp * _iwidth] = _img[xp-shift + yp * _iwidth];
    }
    for (uint32_t xp = x; xp < x + shift; xp++)
    {
      if (_bpp16)
      {
          _img[xp + yp * _iwidth] = (uint16_t) (color >> 8) | (color << 8);
      }
      else
      {
        _img[xp + yp * _iwidth] = (uint8_t)((color & 0xE000)>>8 | (color & 0x0700)>>6 | (color & 0x0018)>>3);
      }
    }
  }
}
```
Example of usage:
```cpp
#include <Arduino.h>
#include <TFT_eSPI.h>

/** TFT_eSPI class for display */
TFT_eSPI tft = TFT_eSPI();
/** Sprite needed for graph scrolling */
TFT_eSprite graph1 = TFT_eSprite(&tft);
/** Sprite needed for graph scrolling */
TFT_eSprite graph2 = TFT_eSprite(&tft);
/** Value for graph1 */
int graph1Val = 1;
/** Value for graph2 */
int graph2Val = 1;

void setup() {
  tft.init();
  tft.fillScreen(TFT_BLACK);
  graph1.createSprite(128, 64);
  graph1.fillSprite(TFT_BLACK);
  graph2.createSprite(128, 64);
  graph2.fillSprite(TFT_BLACK);
}

void loop() {
  graph1.shiftRectLeft(0, 0, 128, 64, 1, TFT_BLACK);
  graph1.drawFastVLine(127,64-graph1Val,graph1Val,TFT_YELLOW);
  graph2.shiftRectRight(0, 0, 128, 64, 1, TFT_BLACK);
  graph2.drawFastVLine(0,64-graph2Val,graph2Val,TFT_RED);

  graph1.pushSprite(0, 32);
  graph2.pushSprite(0, 96);

  graph1Val++;
  graph2Val++;
  if (graph1Val == 65) graph1Val = 1;
  if (graph2Val == 65) graph2Val = 1;
  delay(50);
}
```
This example draws two sawtooth graphs, One scrolling from right to left, the other one opposite.

Bodmer, 2017-12-07:
Hi thanks for the suggestion.

I am currently making a few changes to the library and will build in scroll function.

I have a project that needs vertical scrolling so, to avoid a proliferation of similar functions, the plan is to add two new ones:
`
setScroll(x, y, w, h, [color]); // Sets bounds of scroll woindow, where color is optional, black if not specified

scroll(dx, dy); // dx is -ve/+ve = scroll left/right dt pixels, dy is -ve/+ve scroll up/down dy pixels
`

Bodmer, 2017-12-09:
I clicked the wrong button and closed this by accident.

I am testing an update based on your code and sketch, this will be uploaded soon.

beegee-tokyo, 2017-12-10:
Great! If you need help with testing. let me know.

Bodmer, 2017-12-10:
I have updated the library with scrolling sprites plus a new example called "Sprite_scroll".

I used memmove() as it is quicker that way, plus bounds checking is done to prevent crashes from duff parameters accessing memory areas outside of sprite.

The scroll area is set to the full sprite size upon creation of the sprite
but we can change that by defining a smaller area using "setScrollRect()"if needed:
object.setScrollRect(x, y, w, h, color);
To scroll the sprite:
object.scroll(dx, dy);
Where:
  dx is pixels to scroll, left = negative value, right = positive value
  dy is pixels to scroll, up = negative value, down = positive value

If you can test it for me that would be great. Thanks.

Bodmer, 2017-12-11:
Here is you example modified to work with the new functions:

[code]
#include <TFT_eSPI.h>

/** TFT_eSPI class for display */
TFT_eSPI tft = TFT_eSPI();
/** Sprite needed for graph scrolling */
TFT_eSprite graph1 = TFT_eSprite(&tft);
/** Sprite needed for graph scrolling */
TFT_eSprite graph2 = TFT_eSprite(&tft);
/** Value for graph1 */
int graph1Val = 1;
/** Value for graph2 */
int graph2Val = 1;

void setup() {
  tft.init();
  tft.fillScreen(TFT_BLACK);
  graph1.createSprite(128, 64);
  //graph1.fillSprite(TFT_BLACK); // Optional, filled black by default
  //graph1.setScrollRect(0, 0, 128, 64, TFT_BLACK); // Optional as these are defaults on creation of sprite

  graph2.createSprite(128, 64);
  //graph2.fillSprite(TFT_BLACK); // Optional, filled black by default
  //graph2.setScrollRect(0, 0, 128, 64, TFT_BLACK); // Optional as these are defaults on creation of sprite
}

void loop() {

  graph1.scroll(-1); // Move sprite content 1 pixel left. Default dy is 0
  graph1.drawFastVLine(127,64-graph1Val,graph1Val,TFT_YELLOW);

  graph2.scroll(1); // Move sprite content 1 pixel right. Default dy is 0
  graph2.drawFastVLine(0,64-graph2Val,graph2Val,TFT_RED);

  graph1.pushSprite(0, 32);
  graph2.pushSprite(0, 96);

  graph1Val++;
  graph2Val++;
  if (graph1Val == 65) graph1Val = 1;
  if (graph2Val == 65) graph2Val = 1;
  delay(50);
}
[/code]

beegee-tokyo, 2017-12-11:
works nice and smooth. Tried it in both the example and in my small app with 2 graphs for measured light values. Tried all scroll directions, up, down, left and right and all screen rotations. No problems found.

Tested on ESP32 Dev board with ILI9163 TFT display with 128x128 pixel.

Thanks for implementing.

Bodmer, 2017-12-11:
Thanks for testing.

I have some more checks to do, to verify the functions handle bad/extreme case parameters but it seems robust so far.

Bodmer, 2017-12-14:
OK, all seems good, so closing this one.

closed 2017-12-14 by Bodmer