Bodmer/TFT_eSPI #64: Support for HX8537D?

chaeron, 2017-12-08:
Hi Bodmer:

Any plans to support the newer HX8537D 480x320 TRT displays?  

More specifically, the new Adafruit FeatherWing (https://learn.adafruit.com/adafruit-3-5-tft-featherwing) with ARM-based processors?

Thanks!

Bodmer, 2017-12-09:
I will have a look at the display and see if it can be added. It is probably a case of adding the init code.

The TFT_eSPI library is specifically targeted at the low cost Espressif ESP8266 and ESP32 which are ARM based, however the code is very specific to these processor architectures and the board support packages to optimise performance.  In principle it is possible to adapt the code for any processor, do you have a particular target board in mind.

chaeron, 2017-12-17:
I've been using the Adafruit nRF52 Feather board for my prototypes, but have some ESP8288 and ESP32 processors on order.

JamesNewton, 2018-02-08:
Minor point: Neither the ESP-8266 nor ESP32 are ARM based. The ESP-8266 has a Xtensa Diamond and the ESP32 has a Xtensa LX6, both by Tensilica. The Xtensa system is RISC, but uses less power and is a bit faster than the ARMs.

wawibu, 2018-02-28:
I also plan to switch from 2.2" ILI9341 to 3.5" HX8537D and would be great to still use your library on my ESP8266.
Would that possible to adjust your library in such way?
Thanks

Bodmer, 2018-02-28:
I can create a test version for you but I have no way of testing it myself.  You would have to operate the display in SPI mode.

wawibu, 2018-02-28:
that would be great. plan to use anyhow SPI and my display is right now still on transit. But happy to support you with debugging once I have the display

Bodmer, 2018-03-10:
I have added support for the HX8357D display but I am not able to test it.

Let me know if it works OK!

chaeron, 2018-03-11:
Curious if this will run on an ARM Cortex M0 (Adafruit nRF52) processor?

Bodmer, 2018-03-11:
The library only supports the ESP32 and ESP8266 at the moment. I don't have any short term plans to support ARM processors but that may happen if suitable boards become popular.

closed 2018-03-25 by Bodmer

wawibu, 2018-04-20:
would just like to confirm that HX8357 is working via SPI connected to the ESP8266

Thanks for your support in adding SPI functionality for this driver.

m-menon, 2021-09-23:
> would just like to confirm that HX8357 is working via SPI connected to the ESP8266
> 
> Thanks for your support in adding SPI functionality for this driver.

hi @wawibu , I too have ordered for a couple of boards which use the HX8357. Would you be able to share how you wired the display? I want to be ready by the time my display arrives :-D