Bodmer/TFT_eSPI #66: Touch Buttons and Sliders

jlgoasis, 2017-12-15:
Hi Bodmer, congratulations for your nice library.

Are you thinking to add buttons like switch mode, momentary mode, long press and double press, and basic vertical and horizontal sliders?Otherwise some examples including those?

Thank you in advanced

Bodmer, 2017-12-15:
Hi, thanks for the feedback.

Have a look at [GUISlice ](https://github.com/ImpulseAdventure/GUIslice/wiki)and see if that support library meets your needs, the author recently added compatibility with TFT_eSPI. I have tried the examples and there are different button types and sliders. The library uses the touch screen control functions built into TFT_eSPI.

jlgoasis, 2017-12-20:
Hi Bodmer, again,
Due GUISlice is too complex for a basic Sliders and Buttons aplication (and not very sexy), I've deceided to do my own library based on the TFT_e_SPI_Button class. Unfortunatelly I've detected an incorrect behavior in the justPressed function. If I press this button in my touch screen and I keep it pressed , repeatedly the justReleased button message is shown despite not having stopped touch the display. Please find attached a simple example. I used one of the most standad "Aliexpress" touch display 2.4", 240x320 TFT with the ILI9341 driver.

[TFT_eSPI_On_Off_Button_3.zip](https://github.com/Bodmer/TFT_eSPI/files/1576284/TFT_eSPI_On_Off_Button_3.zip)

Bodmer, 2017-12-21:
OK, I will have a look and see what is happening.

I ported [this library](https://github.com/ginge/GuiLibrary) which hows some promise as it uses callbacks, but there is no slider.

jlgoasis, 2017-12-21:
To give you more details, I want to simuate a Pusch Buttons (momentary funtion). i.e. while the button is pressed, it's ON (I don't like any change to ON/OFF, ON/OFF... as is happening now), and when it's released, just turn it to OFF. Simple....

Bodmer, 2017-12-21:
I don't seem to get the same problem on my touch screens, however I have added some pressure hysterisis and spurious touch pressure reading suppression to debounce the detection in the latest version.

Please test it on you screen. You can optionally change the pressure threshold with a third parameter:
 getTouch(&x, &y, threshold);
Increase the threshold to decrease the sensitivity to touches and increase the pressure hysterisis, values up to 1500 work on my screen, above 20o0 means touches are not detected. Note that the far corners of the screen tend to produce more spurious readings than the middle, this seems to be normal with these low cost touch screens.

Report back it it does/does not work for you.

Bodmer, 2018-01-13:
Closing as I think this is fixed. Let me know if not.

closed 2018-01-13 by Bodmer

Franciscopcosta, 2018-07-23:
hey @jlgoasis did you get this working?
i am having exactly the same issue as you had

jlgoasis, 2018-07-25:
Hi Francisco

May be you speak Spanish like me, I’m from Spain.

Mañana chequeare de nuevo esta función, con el TFT ILI9163, creo que el problema es como la librería o el display gestiona el SPI para el display y para el controlador Touch simultáneamente.

Finalmente decidí usar el TFT RPi touch ILI9486 que es mayor y tiene mejor resolución. Con este display me funciona perfectamente.

Saludos 

Jacinto

fungreenfox, 2022-02-18:
Is there any example on making a circle round touch area?