Bodmer/TFT_eSPI #70: Exception (0): epc1=0x40106921 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

vaderj, 2017-12-25:
When using NodeMCU esp8266 (with ili9341) and recommended config in User_Setup.h + Touch_CS enabled so far all the example sketches work great, but when I flash the Keypad_240x320 example, the following is printed to the serial port and the device constantly reboots:

Exception (0):
epc1=0x40106921 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffef960 end: 3ffefb90 offset: 01a0

>>>stack>>>
3ffefb00:  00000001 00000004 3ffee9c4 402091e3  
3ffefb10:  1fff1fff 00000128 000000c8 402065c0  
3ffefb20:  3ffefb52 3ffefb50 0000ffff 3ffe83c2  
3ffefb30:  00000001 0000000a 3ffee9c4 3ffeeb5c  
3ffefb40:  3fffdad0 00000000 3ffee9c4 40206848  
3ffefb50:  00000000 00000000 3ffee9c4 3ffeeb5c  
3ffefb60:  3fffdad0 00000000 3ffee9c4 3ffeeb5c  
3ffefb70:  3fffdad0 00000000 3ffeeb54 40209e3c  
3ffefb80:  feefeffe feefeffe 3ffeeb70 40100718  
<<<stack<<<
ü!¤„ÿæÊµýþø

Bodmer, 2017-12-27:
Check you have the latest TFT_eSPI version. I have run the example and it works on my setup.

At what point does it crash? Do you get through the touch calibration?

What do you see on the TFT screen when the sketch runs and before it crashes?

Have you connected up the XPT2046 touch controller MISO line to the NodeMCU MISO pin?

Do the Generic On_Off_Button and Touch_Calibrate sketches run OK?

Please provide a web link to the screen you have purchased so I can check the specs. and that it has the right type of touch controller fitted.

eddycyh, 2017-12-27:
I am facing the same issue with esp32. The issue disappear after i changed the following code in drawButton:- 
//static byte tempdatum = _gfx->getTextDatum();
byte tempdatum = _gfx->getTextDatum();

Bodmer, 2017-12-27:
@eddycyh 
Declaring a variable as static will not be the source of the crash, it simply means the variable persists for that function. Please raise a separate issue and post example code that demonstrates the problem you are experiencing.

Bodmer, 2017-12-27:
@vaderj 
I have reproduced the problem you are experiencing with a particular display I have. I will investigate, it appears this problem happens when the XPT2046 is not responding to SPI data requests. This may be because the touch screen connections are reversed or that the chip is faulty.

Bodmer, 2017-12-27:
Ha! The screen [I have is this one](https://i.imgur.com/9kLThRo.jpg). It has the touch controller pin connections and an XPT2046 chip fitted but it has NO touch screen overlay on the display!  This means the XPT2046 returns bad values and eventually the library falls over with a divide by zero error. I will add a check for this condition!

Please check if you have the same problem, post an actual picture of the front and back of your own display for me to check (do not post a link as adverts are often wrong!)

Bodmer, 2017-12-27:
@vaderj
I have updated the TFT_eSPI.cpp file, adding a check for a zero parameter. This should fix your problem.

Thanks for reporting this.  Post back if this does or does not work for you.

closed 2017-12-27 by Bodmer