Bodmer/TFT_eSPI #71: IL9163 128x128 LCD seems to have incorrect CGRAM offsets

vanklompf, 2017-12-27:
IL9163 128x128, or at least version I have doesn't work correctly with library. Screen area is shifted and some parts are not visible.
It seems that CGRAM_OFFSETS in ILI9163_Rotation.h are incorrect for given rotations.
I can see two options here:
1. This is just a bug.
2. This is variant of IL9163 and this option should be parametrized 

I can create PR to fix that, just let me know if this is case 1 or 2.

Bodmer, 2017-12-29:
Unfortunately different manufacturers wire the TFT to map screen pixels to different areas of the CGRAM, so this is probably the issue.  Do you have a simple sketch ( which library example) that demonstrates the problem and I can test on an ILI9163 TFT that I have?

vanklompf, 2017-12-30:
I don't have access to this LCD in this moment. I will have it within few days and I will send sketch and example.

vanklompf, 2018-01-13:
With current library code display looks like that. It is shifted 32 pixels, and part of image is not visible:
[rot0](https://i.imgur.com/lIFRkGh.jpg)
[rot2](https://i.imgur.com/FbJtRTP.jpg)
Screen shifted for all rotation modes.

This is what helps (swapping offsets between different rotation modes)
```
  switch (rotation) {
    case 0:
      writedata(TFT_MAD_MX | TFT_MAD_MY | TFT_MAD_BGR);
      _width  = _width_orig;
      _height = _height_orig;
	  #ifdef CGRAM_OFFSET
        colstart = 0;
        rowstart = 32;
	  #endif
      break;
    case 1:
      writedata(TFT_MAD_MV | TFT_MAD_MY | TFT_MAD_BGR);
      _width  = _height_orig;
      _height = _width_orig;
	  #ifdef CGRAM_OFFSET
        colstart = 32;
        rowstart = 0;
	  #endif
      break;
    case 2:
      writedata(TFT_MAD_BGR);
      _width  = _width_orig;
      _height = _height_orig;
	  #ifdef CGRAM_OFFSET
        colstart = 0;
        rowstart = 0;
	  #endif
      break;
    case 3:
      writedata(TFT_MAD_MX | TFT_MAD_MV | TFT_MAD_BGR);
      _width  = _height_orig;
      _height = _width_orig;
	  #ifdef CGRAM_OFFSET
        colstart = 0;
        rowstart = 0;
	  #endif
      break;
  }
```

I can prepare pull request with offset variants configurable in User_Setup file.

Bodmer, 2018-01-13:
I think being able to set the offsets in the User_Setup file is a good idea. What approach to this doyou suggest?

vanklompf, 2018-01-14:
Yes, this is exactly what I was thinking.
I was thinking something like that
```C++
// There are two variants of 128x128 IL9163. One has CGRAM offset due to unused 32 lines on the beginning of the memory, and one on the end
// If You see that your screen is shifted by 32 pixel switch to other setting
#define CGRAM_OFFSET OFFSET_FROM_START
//#define CGRAM_OFFSET OFFSET_FROM_END
```

I have also other ideas how to tidy up User_Setups and TFT_Drivers:

1. Allow overriding default parameters from user setups. Right now is all or nothing so either use USER_SETUP_LOADED and define everything in your project or use presets. I would like to make it possible to override just selected parameters.
2. Separate responsibilities between USER_Setups and TFT_Drivers i.e define resolution only in User_Setups (I guess there is no use case when only Driver is used, without User_Setup, now it is defined in both which is little bit confusing)
3. Separate public from private in TFT_eSPI.h. Keep in public header only API, and not i.e SPI macros. Create separate header for that.

Let me know if You like those ideas and I can prepare some PRs with proposition of changes.

vanklompf, 2018-01-14:
I have created PR with configuring variants of CGRAM_OFFSET
#76

Bodmer, 2019-01-22:
Closing as I doubt I will update unless more of these displays pop up on the market.

closed 2019-01-22 by Bodmer