Bodmer/TFT_eSPI #72: Recent commit slowed down performance by 3x!!!

chaeron, 2017-12-28:
Hi:

In your commit: 170473b3e4b1c1bc8118cf829e372aeb46721918 on Sunday Dec 17th, Add touch pressure hysterisis and user defined sensitivity, something changed to negatively affect performance by a factor of 3x!

I'm currently testing on an ESP32 dev board with a RPI 480x320 ILI9486-based TFT panel.  I have an automatic testing/debug mode I enable to see how fast my project (a jump rope counter) will run (it uses external sensors which can be time sensitive, so this gives me a baseline of how fast the code will run without sensor delays). The noted checkout slowed down the running of my code by 3x (from a sustained pace of 450 to 138). Reverting to the prior commit with no code changes in my code base speeds things up again.

I think you introduced some significant delays/loops through the hysteresis/sensitivity code!

If you could take a look, that would be much appreciated.  Thanks!!!

Bodmer, 2017-12-29:
Do you have a simple example sketch that you can post to try which demonstrates the problem?

chaeron, 2017-12-29:
Updated the commit ref in the original post.....I cut/pasted the wrong one!

Bodmer, 2017-12-29:
It sounds like you are calling getTouch() in a loop very frequently.

getTouch() uses multiple samples of touch pressure and coordinate checks with sampling delays to improve detection, X+Y coordinate reliability and de-bounce the signal from these very cheap XPT2046 touch based displays. This style of touch screen produces a lot of spurious coordinates so there is quite a lot going on in the "reliability improvement" process and this takes about 10milliseconds for a quick touch and 25ms for a touch and hold.

One way around this is to only call getTouch to obtain the touch coordinates when some touch pressure ("Z" is greater than a threshold )  is detected.  You can do this by including a condition check:

if (tft.getTouchRawZ() > 100)
{
  if (tft.getTouch(&x, &y))
  {
    // Do something here with the detected x,y values
  }
}

In this case the value of 100 is the touch detection pressure threshold, too low and you will get spurious detection, too high and you will have to press really hard on the screen or the touch will not be detected.

getTouchRawZ() if a very "fast" function but may occasionally give a false trigger, but then that will be filtered out by the later getTouch() call.
Report back if this works for you.

chaeron, 2017-12-29:
Been doing some debugging.....turns out it's GUIslice that calls getTouch() 30 times each time through the event loop.  Your hysteresis code bumped a single getTouch() call from 2-3 ms to about 10-15.  Multiply the diff by 30 and it's an extra 300 ms per pass through the loop. Ouch!

I'm looking at the GUIslice code now and will see what I can do to correct the code so that it doesn't call getTouch() 30 times each go round.

Thanks!

closed 2017-12-29 by chaeron

ImpulseAdventure, 2017-12-29:
Hey @chaeron - there is a parameter 'GSLC_MAX_EVT' that controls the maximum number of touch events handled per loop which was provided to permit tuning of responsiveness for certain touch controllers. If you change this to 1 you should see the operation you expect. Sorry that this may have unintentionally appeared like a performance anomaly in TFT_eSPI (which is reliably solid!) and any wasted debug time. I will change the default in the library and mark it with a comment specific to the exception cases instead (or alternately determine a more robust solution for such displays). Thanks again!

chaeron, 2017-12-29:
Thanks Cal....I'll change the parameter and give that a try.

chaeron, 2017-12-30:
Changing the GSLC_MAX_EVT parameter to have a value of 1, in GUIslice, resolve the problem.