Bodmer/TFT_eSPI #73: Exception and reboot

NdK73, 2018-01-08:
Hello.
I'm randomly getting this exception:
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 21 results
0x4020d496: pgm_read_byte_inlined at /home/ndk/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp line 4702 (discriminator 1)
0x40104ef1: ets_timer_disarm at ?? line ?
0x4020d92c: Print::write(unsigned char const*, unsigned int) at /home/ndk/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 38
0x40205fc7: Domotic::handleNet() at /home/ndk/Arduino/libraries/Domotic/src/Domotic.cpp line 237
0x4020bdd5: Print::write(char const*) at /home/ndk/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 187
0x401004d8: malloc at /home/ndk/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c line 1668
0x4020bf70: Print::print(char const*) at /home/ndk/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp line 187
0x40202b88: GUI::setMessage(char const*) at /tmp/arduino_build_822995/sketch/GUI.cpp line 258
0x40230001: pm_onBcnRx at ?? line ?
0x40203450: rccw() at /home/ndk/Dropbox/Applicazioni/Arduino/dispenc/dispenc.ino line 204
0x4020d0ae: std::_Function_handler ::_M_invoke(std::_Any_data const&) at /home/ndk/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 2073
0x402014f2: __analogRead at /home/ndk/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_wiring_analog.c line 33
0x4020233e: std::function ::operator()() const at /home/ndk/Arduino/hardware/esp8266com/esp8266/libraries/ArduinoOTA/ArduinoOTA.cpp line 68
0x40202484: AnEnc_poll() at /tmp/arduino_build_822995/sketch/AnEnc.cpp line 96
0x40203707: loop at /home/ndk/Dropbox/Applicazioni/Arduino/dispenc/dispenc.ino line 135
0x402022b0: _M_manager at /home/ndk/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 1931
0x4020cfe0: _M_invoke at /home/ndk/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/functional line 2069
0x4020cd20: loop_wrapper at /home/ndk/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at /home/ndk/Arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S line 109
```
It sometimes (quite rarely, and usually after some time the system is idle) happens when I touch the encoder, that triggers a call from AnEnc_poll() to rccw() or rcw(). Most of the time it works w/o problems. Might it be something related to font rendering while a network packet is being received? Lately I'm finding the strangest race conditions...

Tks.

Bodmer, 2018-01-11:
It appears that the error is occurring when a byte is read from program memory.

Since this function has most likely been executed before with error this suggests that there is some variable corruption occurring, possible due to a shortage of RAM at some point.

The comment about "race conditions" implies there is instability in the setup so I can only suggest trying to isolate the problem further in your particular setup. The difficulty is that is that debug changes to the code move "move" a problem like this about in the code so you cannot easily identify the root cause. As a starter try to make sure there is plenty of free RAM.

I will leave this issue open for now but I am not convinced that the problem is in the TFT_eSPI library.

NdK73, 2018-01-11:
I doubt it's a RAM shortage. Compiler says I used only 45% of available RAM, and there are 44k free:
```
Lo sketch usa 326103 byte (31%) dello spazio disponibile per i programmi. Il massimo è 1044464 byte.
Le variabili globali usano 37156 byte (45%) di memoria dinamica, lasciando altri 44764 byte liberi per le variabili locali. Il massimo è 81920 byte.
```
(sorry for Italian messages, but the meaning should be clear).
But I checked anyway adding some lines in the setup function:
```
  Serial.println(ESP.getFreeHeap()); delay(10);
```
The minimum value printed is 40024, then crashes.
But I noticed that if I *only* disable the call to analogRead(0) every 5ms (200Hz) it does not crash. Might be some interaction at the SPI level?
With call to analogRead(0) disabled, free heap is always over 38k.

NdK73, 2018-01-12:
Uhm... I tried moving calls to lib directly inside loop() instead of from a sub-sub-sub-routine (actually an indirect callback from a function called from another function called from loop() ) and it seems to work. Could be a stack problem? I still don't know the memory layout of the ESP... Well, now going to look for more problems (using the flash SPI bus to free some IOs)...

Bodmer, 2018-01-12:
OK, thanks for the update.  I will close this for now.

closed 2018-01-12 by Bodmer