Bodmer/TFT_eSPI #74: Operation with alternate SPI0 port?

JamesNewton, 2018-01-11:
The ESP-12 (for example) has pins that access the SPI0 port (normal SPI is SPI1, SPI0 is used in the ESP to access the onboard FLASH) and it /can/ be used in sketches, if, and only if, you allow the ESP to arbitrate via the HWCS pin:
https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#spi

> For this mode the CS will be controlled by hardware as you can’t handle the CS line with a GPIO, you never actually know when the arbiter is going to grant you access to the bus so you must let it handle CS automatically.

As it happens, in my application, I need the SPI1 MOSI pin, D7 / GPIO13, for it's other function, as RX2 via Serial.swap()
https://arduino-esp8266.readthedocs.io/en/latest/reference.html#serial
I don't believe the software serial will work, because I'm also running multiple other services and it has issues:
https://github.com/plerup/espsoftwareserial/issues

Just based on a quick review of the code, it looks like your system expects a GPIO pin for CS and it manages that by itself.  I'm wondering if you think that is really necessary, or could the SPI library mange CS (and therefor leave it to the hardware)... or am I missing something? Honestly, I've not really spent a lot of time on it.

Bodmer, 2018-01-11:
Using SPI0 is supported by the library for the ESP8266, this is called overlap mode. You can see an example setup here.

https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup9_ST7735_Overlap.h

Note that you have to define overlap mode as follows in the setup to tell the library to use the correct port:

#define TFT_SPI_OVERLAP

In overlap mode the NodeMCU pins to use are:

SD0=MISO, SD1=MOSI, CLK=SCLK must use D3 as CS

You can define the DC control pin to the display to be any other pin but in the example setup the pins used are:

#define TFT_CS   PIN_D3
#define TFT_DC   PIN_D5  // Data Command control pin
#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)

Post back if you have any problems.

JamesNewton, 2018-01-12:
Ah! Excellent! Sorry I didn't notice that, as I was searching for SPI0 and overlooked your mention of "overlap" mode in the readme.

The red/green/black tab thing confused me. Mine appears to be a green, but I guess the point is to just try one and if you find things being drawn off the display, try the next one?

So to get started I just edited `User_Setup_Select.h` to un-comment:
` #include <User_Setups/Setup9_ST7735_Overlap.h>`

For now, the issue is:
````
\Documents\Arduino\libraries\TFT_eSPI-master\TFT_eSPI.cpp:189:9: error: 'class SPIClass' has no member named 'pins'

     SPI.pins(6, 7, 8, 0);

         ^

Using library TFT_eSPI-master at version 0.18.10 in folder: 
\Documents\Arduino\libraries\TFT_eSPI-master 
````

I do have NodeMCU 1.0  (ESP-12E Module) selected under tools...

Ah crap... 
https://github.com/esp8266/Arduino/issues/3529
I see the access to SPI0 was only added in ESP core 2.4.0 and I'm 2.3.0... I really hate being on the bleeding edge. Especially since 2.4.0 was a big release. 

I'll look through the change log for 2.4.0 and see if it looks like it's worth the pain...

Bodmer, 2018-01-12:
You may find you only need to update the SPI library in your board support core.
https://github.com/esp8266/Arduino/tree/master/libraries/SPI

I used the Arduino IDE boards manager to change to the latest core version.

JamesNewton, 2018-01-13:
And a few hours later, I've figured out that goes to:
`C:\Users\<username>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SPI`
They seem to have changed that location a few times over the years because there is some seriously out of date information available for google users. The esp8266/Arduino docs hide that, but after looking at them for a while, I'm not sure where to suggest an edit for them... Probably not needed anyway. 

Anywho... Updating those files led to another error, but then I notice the esp8266_peri.h also has some changes to support the SPI overlap mode:
https://github.com/esp8266/Arduino/commit/af0f5ed956b85e41da647d51d1799db469d6e697#diff-faa862c2e8f5c27d08bfb2b40c67a86a
Easy enough to edit in...

And we have a compile! But... nothing showing up on the display.  *sigh*

I'll check over my wiring tomorrow and then pull out the SmartScope and see if I can figure it out.

NdK73, 2018-01-13:
Well, I'm on GIT version and can confirm it's working in shared-SPI mode. Too bad that on breadboard I've had to reduce speed to avoid display corruption... Two extra GPIOs are always useful :)

AcuarioCat, 2018-01-13:
I checked out overlap and it works for me too. Only problem I had was poor response from the touch. I'm using the version from github.com/spapadim/XPT2046 (which I modified for overlap support) rather than the built in one as I couldn't get the built in one to work - didn't detect well and when it did and I set the calibration the results were at best erratic and seemed to be mirrored to where I was touching.

Bodmer, 2018-01-13:
OK, good to hear it is working. I will close this issue.

closed 2018-01-13 by Bodmer

JamesNewton, 2018-01-16:
It's working for me as well now. I had the LCD RESET connect to NodeMCU GND (pin next to D5) instead of D4. ,o) Only issue is that the text is all a mirror image (reversed left/right), but I'm sure that's just some tweaking in the config file. 

Thanks for the help! Much appreciated. Let me know if you ever want a free servo controller, encoder, or DC motor driver kit.

Bodmer, 2018-01-16:
Ah, closed it based on a comment from someone else! Glad to here it is just about there for you. If you cannot fix the mirroring post back.