Bodmer/TFT_eSPI #75: Tools to create RLE fonts

AcuarioCat, 2018-01-13:
Hi, can you tell me the tools you use to generate the RLE fonts.
I want to use font 4 but I need a degrees symbol for temperature. I can replace one of the existing characters (char 27) but I can't find a way of rle encoding my bitmap.
I've checked Irfanview but no support and XnView doesn't seem to do it either.
Any help appreciated!

Bodmer, 2018-01-13:
I write a sketch a few years ago to encode some bitmaps but I am not sure what happened to it!

A character is drawn as a raster scan within a box size of the characters width and height. The RLE format is a byte, the top bit is a flag, the least significant 7 bits are a 7 bit number (0-127). If the flag is 1 then foreground pixels are plotted, if it is 0 the background pixels are plotted. The number of pixels to plot is the 7 bit number +1.

So, for example, the encoded sequence "0x81, 0x1F" means plot 2 pixels in foreground then 32 pixels in background.

You could manually generate a single character using some squared paper and a pencil and replace the one in the file.

Bodmer, 2018-01-13:
I found I replaced the Grave "`" accent with a degree symbol in that font some time ago, ascii code 96 (decimal). Try that.

AcuarioCat, 2018-01-13:
I worked it out in the end reverse engineering the code (phew..). For anyone trying to do the same remember to change the value in the width table for the width of the new character. Mine is a bit different to yours and ended up as follows:

//Changed to degree symbol for temperature display
PROGMEM const unsigned char chr_f32_27[] = 
{
0x17, 0x82, 0x04, 0x86, 0x02, 0x81, 0x02, 0x81,
0x01, 0x81, 0x04, 0x81, 0x00, 0x81, 0x04, 0x81,
0x00, 0x81, 0x04, 0x81, 0x01, 0x81, 0x02, 0x81,
0x02, 0x86, 0x04, 0x82, 0x7f, 0x1a
};

closed 2018-01-13 by AcuarioCat