Bodmer/TFT_eSPI #78: tft.fontHeight() and tft.textWidth() return double values on /second/ call.

JamesNewton, 2018-01-19:
I'm running into a really interesting issue and I /think/ it's not me doing something stupid, so I'm going to report it and see if anyone else can reproduce it.

When you call tft.fontHeight() or tft.textWidth() with a given string and font number (just font for Height) you get correct values, the FIRST time. When you call them again, they provide DOUBLE that value. 

Sadly, this seems to only be happening in MY sketch.  All the example sketches seem to work fine, and the TFT_Print_Test sketch also reports the width and height correctly when modified to do so. I'm slowly working through adding things to the basic TFT_Print_Test sketch to see when it will start happening again. 

````
/*  
Mockup of status panel.
Tools / Board = NodeMCU 1.0 (ESP-12 Module)

https://www.adafruit.com/product/2088
https://github.com/Bodmer/TFT_eSPI
https://learn.adafruit.com/adafruit-gfx-graphics-library?view=all

 */

#include <TFT_eSPI.h> // Graphics and font library for ST7735 driver chip
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h
String reading;
#define TFT_GREY 0x5AEB // New colour
#define TFT_RBG_COLOR(r,g,b) ((r << 11) | (g << 5) | (b))
ADC_MODE(ADC_VCC);

void setup(void) {
  tft.init();
  tft.setRotation(1);
  randomSeed(analogRead(0));
}

void showreading(String reading, int x, int y, int fontno) {
  int x1 = x;
  int x2 = (tft.textWidth(reading,fontno));
  tft.fillRect(0,y,TFT_WIDTH,tft.fontHeight(fontno)/2,TFT_BLACK);
  if (x<0) { 
    x1 = -x; // do this in one operation and x ends up negative again somehow.
    x = x1-x2;
    }
  tft.setTextColor(TFT_YELLOW,TFT_BLACK); 
  tft.setCursor(0, y, fontno); tft.setTextSize(1);
  tft.print(x);//reading);
  tft.print(" ");
  tft.print(reading.length());
  tft.print(" ");
  tft.print(x2); // PROBLEM: This shows 90 when it should show 45 on the second and following loops
  }
  
void loop() {
  // Fill screen with grey so we can see the effect of printing with and without 
  // a background colour defined
  tft.fillScreen(TFT_BLUE);
  //tft.fillRect(0,0,TFT_WIDTH,TFT_HEIGHT,TFT_BLACK);  delay(200);
  
  // (cursor will move to next line automatically during printing with 'tft.println'
  //  or stay on the line is there is room for the text with tft.print)
  tft.setCursor(0, 2);
  tft.setTextColor(TFT_YELLOW,TFT_BLACK); tft.setTextFont(1); tft.setTextSize(1);
  tft.println("SSID:MassMind.org");
  tft.println("PASS:192.168.4.1");
  
  // Set the font colour to be red with black background, set to font 4 //4 is too big, "F" wraps.
  tft.setTextColor(TFT_WHITE,TFT_BLUE);    tft.setTextFont(1); tft.setTextSize(1);
  tft.println("Line 1");
  tft.println("Line 2");
  tft.println("Line 3");
  tft.println("Line 4");
  tft.println("Line 5");
  tft.println("Line 6");
  tft.println("Line 7");
  tft.println("Line 8");
  tft.println("Line 9");
  tft.println("Line 10");
  tft.println("Line 11");
  tft.println("Line 12");
  tft.println("Line 13");
  tft.println("Line 14");

  delay(2000);
    
  //while(1) yield(); // We must yield() to stop a watchdog timeout.

  for (int i = 50;i>0;i--) {
    reading = String((float)random(100000)/1000,3);
    showreading(reading,-100,TFT_HEIGHT/4,2);
    tft.setTextColor(TFT_WHITE,TFT_BLUE); tft.setTextFont(1); tft.setTextSize(1);
    tft.setCursor(0, (TFT_HEIGHT-tft.fontHeight(1)));
    tft.print("Line ");
    tft.print(max(abs(-100)-tft.textWidth(reading,4),0));
    tft.println("   ");
    tft.setCursor(0, 2);
    tft.setTextColor(TFT_GREEN,TFT_BLACK); tft.setTextFont(2); tft.setTextSize(1);
    tft.print("192.168.200.");
    tft.setTextFont(1); tft.setTextSize(2);
    tft.println("273");
    tft.drawLine(0, 0, 128, 0, TFT_BLACK); //erase the prior reading
    tft.drawLine(0, 0, (int)(ESP.getVcc()/50), 0, TFT_BLUE);// TFT_RBG_COLOR(16,0,16));
    tft.drawLine(0, 1, TFT_WIDTH, 1, TFT_BLACK); //erase the prior reading
    tft.drawLine(0, 1, (int)random(80), 1, TFT_RBG_COLOR(30,0,27));

    delay(500);
    }
}

```

Bodmer, 2018-01-20:
tft.setTextSize(2); is a font height and width multipier. After drawing text at double height you need to set it back to tft.setTextSize(1); otherwise returned values will be doubled.

JamesNewton, 2018-01-21:
*facepalm*

closed 2018-01-21 by JamesNewton