Bodmer/TFT_eSPI #82: slower than 16.15?

MarcFinns, 2018-01-27:
Updating from 16.15 i had also to update several functions such as renderJpeg, as apparently pushcolo has a different prototype. in any case I used your functions. However the result is way slower than it used to be. Possible...? I am using an ESP8266. thanks again

Bodmer, 2018-01-27:
Can you provide a simple sketch that illustrates the problem?

Have you tired the 160x128 example TFT_flash_jpg and does it run slow?

Bodmer, 2018-01-27:
Which pushColor(s) function was incompatible?

MarcFinns, 2018-01-27:
Hi Bodmer, hard to extract a sketch as it is integrated in a much larger project 
[LINK](https://www.youtube.com/watch?v=iyFuKU8ZcuA). Have a look at the video. Everything is much slower now. 
Among one of the features there is a modified planespotter based on your version, so i guess you could see it from there? Specifically, the older function 
 void GfxUi::renderJPEG(int32_t xpos, int32_t ypos)
would not compile the line _tft->pushColors(pImg, mcu_pixels);    // pushColors via 64 byte SPI port buffer

I replaced it with a more recent version as below and it works. However, everything is slower including text only screens, not only this specific function.
renderJPEG(int xpos, int ypos) 
Some details on the configuration:
CPU is ESP8266 at 80MHz.
Clock is boosted momentarily to 160MHz when handling heavy tasks such as JPEG decoding.
The display is an ILI9341. To save on I/O, Reset is attached to CPU reset, CS is at GND, DO as DC and SPI Frequency is 27 MHz.
Please let me know how I can help to pinpoint the issue… Thanks again.

Bodmer, 2018-01-27:
I've reinstated the 8bit pushColors() function.

Performance tests on the weather station code for an example jpeg indicated:
98ms with re-instated pushColors 8 bit fn
96ms with new pushColors 16 bit fn
143ms with while loop and unbuffered pushColor 16 bit (expected to be slower)

Code used for tests:

#ifdef USE_SPI_BUFFER
      // Now set a MCU bounding window on the TFT to push pixels into (x, y, x + width - 1, y + height - 1)
      _tft->setWindow(mcu_x, mcu_y, mcu_x + win_w - 1, mcu_y + win_h - 1);
      // Write all MCU pixels to the TFT window
      ////uint8_t *pImg8 = (uint8_t*)pImg;     // Convert 16 bit pointer to an 8 bit pointer
      ////_tft->pushColors(pImg8, mcu_pixels*2); // Send bytes via 64 byte SPI port buffer
      _tft->pushColors(pImg, mcu_pixels, false); // New alternative to above 2 lines <<<<<<<<<<<<
#else
      // Now set a MCU bounding window on the TFT to push pixels into (x, y, x + width - 1, y + height - 1)
      _tft->setAddrWindow(mcu_x, mcu_y, mcu_x + win_w - 1, mcu_y + win_h - 1);
      // Write all MCU pixels to the TFT window
      ////while (mcu_pixels--) _tft->pushColor(*pImg++);
      _tft->pushColors(pImg, mcu_pixels); // Alternative to above line<<<<<<<<<<
#endif

closed 2018-01-27 by Bodmer

MarcFinns, 2018-01-27:
Hi Bodmer, thanks for taking the time. 
Is this explaining the general perceived reduction in performance?
Does it makes sense / help if i rebuild the old firmware and add performance logs?

MarcFinns, 2018-01-27:
I am fine peerformance-wise with your latest modification. On a related subject since you tested the planesp[otter, do you see also a terrible slowness in the adsbclient today, or is it a problem of my refreshed code...? thanks again.

Bodmer, 2018-01-27:
I have not noticed anything unusual today but there are times when the ADSB server seems slow to respond or does not respond to every request.  I am using 2.4.0-rc2 core which does seem slower than 2.3.0 at times but that is just a perception during testing of code and may not be the case!

MarcFinns, 2018-01-28:
Indeed. Now back to normal. Thanks