Bodmer/TFT_eSPI #83: ENHANCEMENT PROPOSAL - softwareReset()

MarcFinns, 2018-01-27:
Hello Bodmer, I propose an enhancement that i originally implemented myself in an earlier version. 
I exposed an additional function **_softwareReset()_** that only sends the SWRST and the init sequence to the LCD, keeping everything else intact in the driver (e.g. rotation etc).
This is very useful at least in some of my projects. For example, when turning on the LED backlight via proximity sensor in a situation where the battery is almost depleted, the step-up DCDC converter might not be fast enough to react. The only component that has a problem with it is the screen, that goes totally white. Calling that function in the process of turning on the screen works around the issue. Not sure if it could be useful in other situations, but it is very easy to implement hence perhaps it could go in the master codebase?
thanks

Bodmer, 2018-01-27:
Sounds like a software work-around for a hardware problem!

You can call tft.begin(); at any time, not just in setup() and that will re-initialise the TFT. You will meed to call  tft.setRotation(x); afterwards though to set the rotation as you desire. You can call tft.getRotation() if you do not have a record of the current rotation. tft.setRotation(tft.getRotation()); will work too as the libraries rotation record is not changed during the TFT reset.

Since the feature is already there I do not see a need to make any changes.

closed 2018-01-27 by Bodmer

MarcFinns, 2018-01-27:
Well, it is not unusual to write software around the hardware edges, ask Intel these days... ;)
Calling init/begin is not exactly the same though. Begin() re-initialises plenty of other things on top of the display, including SPI which is a problem. 
In cases where MISO is not used for the display (a valid use-case) and the related GPIO drives something else, it creates a mess.

Bodmer, 2018-01-27:
OK, fair point. I have changed the code so the pins only get set after bootup and not when called later in a sketch. If a hardware reset is connected this is the preferred approach as I found that a TFT can get in a state that ignores a soft reset, so the code uses the hardware reset if connected instead.

Calling setRotation() after an init makes sense too as this will then invoke and offsets.

I have uploaded the change, tft.init() can now be called safely in a sketch.

MarcFinns, 2018-01-27:
great, thanks!!!