Bodmer/TFT_eSPI #84: Different behaviour detected - wrap to top

MarcFinns, 2018-01-27:
Hi Bodmer, sorry for spamming.
I detected a new behaviour that is different from the previous 16.15 version i was using.
In the old version, when using println(), if the bottom of the screen was reached the printout simply disappeared. In the new version, it wraps to the top.
Which one is the intended behavior? Is there a way to revert to the previous behavior?
thanks again!

Bodmer, 2018-01-27:
Thanks for your inputs, I do not regard it s spamming at all!

I added the y wrap for my project and I regarded the lack of it as a bug! I have now disabled it is the latest upload from a few minutes ago.

You can stop wrapping on the x limit by using tft.setTextWrap(false) but that did not stop a y wrap in some perhaps undesirable circumstances.

I will add a switch on the y wrap function at a future date.

closed 2018-01-27 by Bodmer

MarcFinns, 2018-01-27:
great, thanks!